adafruit/circup #1: no output when running

ladyada, 2019-09-02:
hiya i tried running `python circup.py list` and there's no output. i tried on win10 and mac OS with the feather m0 plugged in and some libraries on it

[image]

please add output to tell me if a board was detected or not, or what point it stopped :)

ntoll, 2019-09-03:
A heinous oversight... I'll get this fixed ASAP. ;-)

ntoll, 2019-09-03:
Fixed by a2f81f90821.

[image]

ladyada, 2019-09-03:
still nothing on windows, nothing logged either

[image]

[image]

ladyada, 2019-09-03:
[image]

ntoll, 2019-09-03:
Hmm... that's weird. I'm running this on Linux. Will test on Windows tomorrow and fix. Re-opening until it's fixed (the `click` library I'm using for this is supposed to be cross platform).

ntoll, 2019-09-04:
OK... I've been trying things out on my Windows 10 laptop. Frustratingly, it works for me (see attached screenies). :-/

Apologies for asking the simple/obvious questions, but I just want to be sure that everything is set up correctly on your version of Windows:

* You're using a virtualenv
* With >= Python3.5
* When you do a `git fetch` or `git pull origin master` to get the latest, do you also run `pip install -e ".[dev]"` to ensure you've got all the necessary dependencies? (Easily forgotten)
* By using the `pip` command to "install" the code in the current virtualenv we're able to use the `circup` utility in exactly the same way as the end user... i.e., there's a `circup` command added to the user's path (the `-e` flag tells pip to install the package in "edit" mode from the code within the current directory). What's the output of just `circup`? (I see the help information).

Once again, sorry for the "department of the obvious" type questions, but I'm anxious that this works across all platforms and painfully aware (Hello Mu) that what works for me, may not work for others. Care to share version information too?

Here's the screenies.

Running the `circup list` command:

[image]

The resulting logs:

[image]

ntoll, 2019-09-04:
Another thought... could this be down to the type of device that's connected..? I'm testing with a CPX. As far as I understand it though, all CIRCUITPYTHON identified devices work in the same way when it comes to the flash file system (from the OS's point of view).

ntoll, 2019-09-04:
Hi @ladyada 

I've just pushed a bunch of work from today, I've just tested it on my Windows 10 laptop and it all works as expected for me. Please can you test and let me know know you get on. E.g.

```
$ circup
Usage: circup [OPTIONS] COMMAND [ARGS]...

  A tool to manage and update libraries on a CircuitPython device.

Options:
  --verbose  Comprehensive logging is sent to stdout.
  --version  Show the version and exit.
  --help     Show this message and exit.

Commands:
  freeze  Output details of all the modules found on the connected...
  list    Lists all out of date modules found on the connected
          CIRCUITPYTHON...
  update  Checks for out-of-date modules on the connected CIRCUITPYTHON...
```

All feedback most welcome..! :+1:

You should see something like the following:

[image]

ntoll, 2019-09-25:
I'm assuming this is fixed now, so closing. Please don't hesitate to re-open otherwise.

closed 2019-09-25 by ntoll