adafruit/circup #6: Minimum supported version of Python?

ntoll, 2019-09-03:
@limor I'm assuming Python 3 (specifically Python 3.5 or 3.6 -- which one?) as the minimum Python requirement for the `circup` utility. 

While I'm happy to make appropriate changes so this works with Python 2, I'm not sure it's really useful nor helpful since Python 2's EOL is 1st January 2020 (less than four months away). ;-)

ntoll, 2019-09-05:
Sorry cc/@ladyada (doh)

ladyada, 2019-09-05:
hiya no support for py2 needed. 3.x is fine, 3.5 is totally great, we see that version a lot :)

ntoll, 2019-09-06:
Coolio. I've tested the tool with Python 3.5, 3.6 and 3.7 on Linux (Ubuntu 18.04), Windows (10) and an older version of OSX (10.13.6).

A good next step would be to add some CI to automate this via Travis or Appveyor or some other such service. Also, automatic ReadTheDocs integration would be useful too.

I'll close this issue. @ladyada could you please confirm it works on Windows for you (#1).

closed 2019-09-06 by ntoll

ladyada, 2019-09-06:
yes confirmed it works with windows (or at least, i get correct-lookin' output!) will do more testing later :)

ntoll, 2019-09-06:
Hurrah!