adafruit/circup #8: Extracting __version__ value from .mpy

dhalbert, 2019-09-03:
If you run `strings` on a .mpy, you can find the version number close to "__version__". It seems to be always just before, but I haven't done an extensive check, e.g.
```
$ strings neopixel.mpy
...
neopixel_write
neopixel_write
3.3.7
__version__
__repo__
...
```

ntoll, 2019-09-04:
OK... so I've been thinking about this. I wonder if you could just use the REPL to:

* Restart the device.
* import the module.
* print the `__version__` and `__repo__` metadata via the serial connection.

Rinse and repeat for all *.mpy modules to capture the data needed.

Thoughts..?

ntoll, 2019-09-06:
@dhalbert also, your suggestion requires ``strings`` to be installed (not necessarily guaranteed -- I'm not even sure it's out-of-the-box in Windows).

dhalbert, 2019-09-06:
Importing the module might be OK, though it requires resetting the board at some point to clear that out.

I wasn't necessarily thinking of using `strings` itself, but instead finding `__version__` in the .mpy and then looking nearby for something that looks like a version string. Could use a Python regexp over the whole .mpy file for that.

ntoll, 2019-09-06:
Ack. 

I guess the problem is simply, how do you parse a `.mpy` file? Regex could work, but Python definitely does! ;-)

(Also, thinking out loud here, we'd need to connect to the device via serial, in order to extract the version details for CircuiPython running on the board, to ensure the correct compiled version of the bundle is downloaded.)

dhalbert, 2019-09-06:
There is a tool in micropython/CircuitPython, `mpy-tool.py` which is at least a partial decompiler of .mpy files. I'm trying to get it to work and will see if it can dump out what you need.

dhalbert, 2019-09-06:
The purpose of `mpy-tool.py` is generate compilable C files for frozen modules, so it reads .mpy files and extracts the info. In particular, it generates a table like this, for example:

from neopixel.mpy:
```
extern const qstr_pool_t mp_qstr_const_pool;
const qstr_pool_t mp_qstr_frozen_const_pool = {
    (qstr_pool_t*)&mp_qstr_const_pool, // previous pool
    MP_QSTRnumber_of, // previous pool size
    74, // allocated entries
    74, // used entries
    {
        (const byte*)"\xbd\x08" "<module>",
        (const byte*)"\xae\x0b" "neopixel.py",
        (const byte*)"\x35\x04" "math",
        (const byte*)"\x19\x09" "digitalio",
        (const byte*)"\x2b\x0e" "neopixel_write",
        (const byte*)"\x12\x05" "3.3.7",
        (const byte*)"\x3f\x0b" "__version__",
        (const byte*)"\x8d\x08" "__repo__",
...
```
It relies on reading other files to figure out how to parse a .mpy, but it could be stripped down to something much simpler that would output the probable version string. (I would even volunteer to do that if you wanted to use this technique.)

This "qstr" table is a table of interned strings. The URL from `__repo__` is too long, so it's not in that table, but it appears later in the file. Full output here:

[neopixel.mpy.frozen-c.txt](https://github.com/ntoll/circup/files/3585788/neopixel.mpy.frozen-c.txt)

ntoll, 2019-09-23:
OK... After email conversation with PT here's the approach to be taken:

```
We prefer the brute-force parsing method for now instead of the elegant but
time-consuming "fully attempt to parse mpy format" technique.

Please add this capability by checking if the file ends in .mpy and doing some binary 
parsing dance - looking for __version__ in the bytearray, then backtracking to find the
valid semver, it seems strings are prefaced by a byte-length marker :)
```

ntoll, 2019-09-23:
See PR #12. Feedback most welcome..! :slightly_smiling_face:

commit b2a94e1c (2019-09-23): Handle .mpy based modules. Fixes #8.
closed 2019-09-23 by ntoll