adafruit/circup #9: Use bundle as ground truth version information instead of querying GitHub?

dhalbert, 2019-09-03:
Instead of making many queries to Github, an idea is to download the latest bundle (could use the .py version for ease), and extract the version information from that. That would make any updates consistent with what's in the latest bundle, and save on GitHub queries.

ntoll, 2019-09-03:
Good idea. The API request limit is something that's causing a bit of a problem. This should also simplify the internal logic too (rather than having to process a number of requests to GitHub). I'll see what I can do tomorrow. :+1:

sommersoft, 2019-09-03:
>  (could use the .py version for ease)

There's also the `VERSIONS.txt` file available in the bundle. Like the files themselves, the version is build dynamically during bundle building. Could easily RegEx it into a dict:
```
20190815
https://github.com/adafruit/Adafruit_CircuitPython_74HC595/releases/tag/1.0.2
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/releases/tag/2.1.0
https://github.com/adafruit/Adafruit_CircuitPython_ADT7410/releases/tag/1.0.0
https://github.com/adafruit/Adafruit_CircuitPython_ADXL34x/releases/tag/1.10.2
https://github.com/adafruit/Adafruit_CircuitPython_AM2320/releases/tag/1.1.3
```

ntoll, 2019-09-03:
@sommersoft this is gold dust. Thank you. 

Up until now I was parsing whatever was in `__version__`, and (obviously) this was failing for code I was checking hosted on here since the `__version__` is always `0.0.0-auto.0` (or something like it). 

Finding a single shot "canonical" version of the latest versions is my current "philosophers stone". You folks know these repos better than I, so please don't hesitate to make suggestions. :+1:

sommersoft, 2019-09-03:
As this thought-train completed, using VERSIONS.txt might not work out all that easily. File/package names _might_ not match the repo name. I'll try to give some possibilities a run through...

ntoll, 2019-09-03:
@sommersoft I have access to the `__repo__` metadata though, so I should be able to make a match.

dhalbert, 2019-09-03:
We could modify `VERSIONS.TXT` to include the module name as well.

ntoll, 2019-09-04:
OK... so I've been playing around with your suggestion to download the bundle as a single download. Let's just say it works a LOT quicker than all those queries for each individual module to the GitHub site. The logic will be something like:

* Check if there's a local copy of the bundle (with tag data).
* Compare to current latest.
* If there's a later tag, update.
* ... etc etc etc do fun stuff to update versions on plugged in device...

I've spiked up something like the above. Now I'll turn it into something proper with tests, docs etc... ;-)

ntoll, 2019-09-04:
OK... so I've been playing around with places to get versioning metadata. Here's what I've found:

* it looks like the GitHub URLs in VERSIONS.txt are truncated (at least, in the current latest bundle; see attached) :-(
* I can use the same metadata extraction technique as I use on the modules on the device on the modules in the downloaded bundle.
* If there could be a VERSIONS.json as part of the bundle, that would be wonderful. Something like...

```
[
  {
    "project": "adafruit_xyz",
    "version": "1.2.3",
    "repo": "https://github.com/adafruit/adafruit_xyz.git"
  }
  ... etc ...
]
```
[VERSIONS.txt](https://github.com/ntoll/circup/files/3574503/VERSIONS.txt)

ntoll, 2019-09-04:
@ladyada do you have any +/- opinions on this approach. I like it because it's quicker since only a couple of network calls are needed and the whole bundle takes < 1 to download (for me).

ntoll, 2019-09-04:
I should add only *some* of the repositories are truncated. E.g. the penultimate item is:

```
https://github.com/adafruit/Adafruit_CircuitPython_tu/releases/tag/1.1.0
```

sommersoft, 2019-09-04:
hmm. There's definitely a bug there. I count three truncated URLs; the `tu` (i'm assuming its `turtle`), and two that are simply `CircuitPython_/`.

I ran a quick debug, and it has to do with how the repos are listed in the `.gitmoduels`. The three in question don't have `.git` appended to the remote URL. Easy fix.

ntoll, 2019-09-04:
@sommersoft :+1: FWIW the current version avoids VERSIONS.txt and walks the `lib` directory to find files with the expected metadata.

ntoll, 2019-09-04:
I've spent today taking this approach and it works really well. Hardly any need to ping GitHub thus mitigating the throttling problem I'd encountered. This also significantly speeds up the command.

See: 2a7c72bc

[image]

ntoll, 2019-09-05:
I'll close this, since it's the approach that works and I've documented the modus operandi and advantages of this approach in the README, here: https://github.com/ntoll/circup#how

As always, I'm more than happy to re-open, change things or explain further. Just ping me..! :-)

closed 2019-09-05 by ntoll