adafruit/circup #17: Installing from PyPI errors

makermelissa, 2019-10-04:
Typing `pip3 install circup` results in the following error:

```
Collecting circup
  Downloading https://files.pythonhosted.org/packages/67/94/fb0526baa550ddfc503b49edbe4932c7e1999e8aa1488cc30f1057cc0547/circup-0.0.1.tar.gz
    ERROR: Complete output from command python setup.py egg_info:
    ERROR: Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/5h/13ngwp912tq9gr12796g77br0000gn/T/pip-install-6xwg6yoc/circup/setup.py", line 21, in <module>
        with open(os.path.join(base_dir, "CHANGES.rst"), encoding="utf8") as f:
    FileNotFoundError: [Errno 2] No such file or directory: '/private/var/folders/5h/13ngwp912tq9gr12796g77br0000gn/T/pip-install-6xwg6yoc/circup/CHANGES.rst'
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/5h/13ngwp912tq9gr12796g77br0000gn/T/pip-install-6xwg6yoc/circup/
```

siddacious, 2019-10-04:
when I downloaded the files from pypi, there's a bunch of stuff missing when compared to the repo:
[image]

I'm not familiar enough with PyPi to know if it prunes things, but importantly it complains about `CHANGES.rst` being missing:
[image]

ladyada, 2019-10-04:
@ntoll hiya we're tring to automate installation to pypi on release but having an issue here - do you have any suggestions? thanx!

sommersoft, 2019-10-04:
I can get `CHANGES.rst` to be added to `egg-info/SOURCES.txt` by using a manifest. Looking at the Travis deploy log, this seems to be the issue. When the egg info and sdist are created, it doesn't include `CHANGES.rst` so PyPI doesn't grab/distribute it.

A secondary issue, is the `__version__` string. As this is configured, it must be changed by hand in `circup.py`, otherwise PyPI will always replace the current version.

I'll push a quick PR, with `__version__ = 0.0.3` to avoid that for now.

makermelissa, 2019-10-04:
It installed correctly this time and version looks good. 🌮

closed 2019-10-04 by makermelissa

ntoll, 2019-10-07:
Hi Folks,

Apologies for the silence on my part, just been catching up today after a few hectic days.

My advice is to just use the Makefile to update the package on PyPI. See:

```
make check - run all the checkers and tests.
make dist - make a dist/wheel for the project.
make publish-test - publish the project to PyPI test instance.
make publish-live - publish the project to PyPI production.
```

The `publish-*` commands always run `check` and `dist` and will abort if they encounter any problems (so you can't upload broken packages).

Also, I *strongly* suggest you use the PyPI test instance (and `make publish-test`) to check everything works OK both in terms of uploading the assets and testing they're installable (see https://packaging.python.org/guides/using-testpypi/ for details of how this actually works).

Does this help..?

My Fridays and weekends are usually quite hectic (Friday = London with young coders, weekend = family times), so in case of urgency ping me via discord (I have an alert configured to email me whenever I'm @mentioned).