adafruit/circup #19: Automate Version So GitHub & PyPI Always Match

sommersoft, 2019-10-04:
The PyPI [version info](https://github.com/adafruit/circup/blob/master/setup.py#L63) is determined by the value of `__version__` in `circup.py`.

If this isn't updated in a PR **prior** to a release on GitHub, the PyPI deployment will simply overwrite the annotated (previous) version.

ntoll, 2019-10-07:
@sommersoft yup :+1:

How about a script that's kicked off by `make publish-*` that checks the current version on the remote instance of PyPI and complains if the versions don't align properly (i.e. the local version MUST have a higher semver value than the current version in PyPI)..? This is not a difficult job.

makermelissa, 2020-04-17:
I just added Github Actions in place of Travis, so perhaps that might be useful. I left the versioning manual though.

askpatrickw, 2020-12-18:
working on this in #46

askpatrickw, 2020-12-22:
This is fixed by PR #46

closed 2020-12-22 by askpatrickw