adafruit/circup #20: Error while running CircUp Show during bundle release

makermelissa, 2019-10-29:
I kind of got lucky finding this one. At the time I ran it, the release had been created, but the bundle asset hadn't been finished being generated.

```
bash-3.2$ circup show
Found device at /Volumes/CIRCUITPY, running CircuitPython 5.0.0-alpha.4.
Downloading latest version information.

Traceback (most recent call last):
  File "/usr/local/bin/circup", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/circup.py", line 719, in show
    available_modules = get_bundle_versions()
  File "/usr/local/lib/python3.7/site-packages/circup.py", line 403, in get_bundle_versions
    ensure_latest_bundle()
  File "/usr/local/lib/python3.7/site-packages/circup.py", line 505, in ensure_latest_bundle
    get_bundle(tag)
  File "/usr/local/lib/python3.7/site-packages/circup.py", line 547, in get_bundle
    r.raise_for_status()
  File "/usr/local/lib/python3.7/site-packages/requests/models.py", line 940, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/download/20191029/adafruit-circuitpython-bundle-py-20191029.zip
```

ntoll, 2019-10-29:
Nice find @makermelissa..! :+1:

I'm happy for you to argue against this, but I'll close it as "won't fix" because this is more to do with the "alignment of planets" (if you see what I mean) than any buggy code. I believe it a sensible assumption for the software to make that the assets are all correctly updated/synced on GitHub.

Sometimes other users may encounter this problem for the reasons you state, but when they Google for the error, they'll find this bug report and my answer (and they'll also hopefully figure out to wait a minute or two, while GitHub re-syncs with the new release).

Also, "hello posterity". ;-)

makermelissa, 2019-10-29:
I was thinking maybe handle the error a little nicer like letting the user know to wait a few moments. I also imagine this affects anything else that reads the bundle like install.

ntoll, 2019-10-30:
Yes, I woke up this morning and thought the same. There should be a message that suggests that they wait for GitHub to sync. I'll get this fixed up by the end of this week (it's school holidays here in the UK so full on family things happening at the moment).

ntoll, 2019-10-30:
@makermelissa okedokey... I've just added a fix in the PR. I agree with you that this is a much better solution than just ignoring it. ;-)

Once you're happy with the PR, please merge and close this issue.

Thank you for reporting this and giving me a kick to make `circup` more friendly for users.

closed 2019-10-31 by makermelissa

ladyada, 2019-10-31:
thank you @ntoll !

ntoll, 2019-10-31:
NP :-)

PR #21 (2019-10-31): Handle HTTP Error codes when ensuring bundle update.
Fixes #20.

This PR introduces two changes:

* If an HTTP error code is encountered when updating the bundle release, then log the error, print a friendly error message and sys.exit 1 (error).
* Tests for the above.

Thanks to @makermelissa for reporting this problem. :+1:
commit 75e4ba52 (2019-10-31): Fixes #20.