adafruit/circup #39: Errors (possibly due to upgrading to Python 3.9

mattpackwood, 2020-10-10:
I just upgraded to Python3.9 and am getting the following issue:

```
circup update --all
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pkg_resources/__init__.py", line 582, in _build_master
    ws.require(__requires__)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pkg_resources/__init__.py", line 899, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pkg_resources/__init__.py", line 790, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (click 7.1.2 (/Users/mattpackwood/Library/Python/3.9/lib/python/site-packages), Requirement.parse('Click==7.0'), {'circup'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/System/Volumes/Data/Users/mattpackwood/Library/Python/3.8/bin/circup", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3257, in <module>
    def _initialize_master_working_set():
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3240, in _call_aside
    f(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3269, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pkg_resources/__init__.py", line 584, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pkg_resources/__init__.py", line 597, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pkg_resources/__init__.py", line 785, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'Click==7.0' distribution was not found and is required by circup
```

askpatrickw, 2020-10-11:
@mattpackwood  run `pip list`, is Click also installed?

ntoll, 2020-10-12:
This is the key line:

`pkg_resources.DistributionNotFound: The 'Click==7.0' distribution was not found and is required by circup`

It looks like the `Click` command probably needs to be pinned to a later version than `7.0` (which probably isn't packaged for Python 3.9, hence the error).

I quickly created PR #40 as a potential fix.

mattpackwood, 2020-10-12:
I ran:

`Matt-Packwoods-iMac:~ $ python3 -m pip list | grep click`
And got:
`click                                             7.1.2`

Matt

askpatrickw, 2020-10-12:
See https://github.com/pallets/click/issues/1684 which was fixed and merged https://github.com/pallets/click/pull/1685
But I don't think is released yet since 7.1.2 (latest) was released in April.

askpatrickw, 2020-10-12:
Sorry.. they reverted the fix but did not re-open the issue. I let them know.

askpatrickw, 2020-10-16:
I had not actually used circup and thought I'd give it a shot. To work around Click 7x not supporting Py 3.9.90 I installed from GitHub: `pip install git+https://github.com/pallets/click.git`.

**Environment**  

```cli
code/local/circuit-python-scripts via 🐍 cp_base
➜ pip list
Package    Version
---------- ----------
appdirs    1.4.4
certifi    2020.6.20
chardet    3.0.4
circup     0.2.2
click      8.0.0.dev0
idna       2.10
pip        20.2.3
requests   2.24.0
semver     2.10.2
setuptools 49.2.1
urllib3    1.25.10

code/local/circuit-python-scripts via 🐍 cp_base
➜ python --version
Python 3.9.0
```

Anyway, at this point, I get logger errors:

```python
--- Logging error ---
Traceback (most recent call last):
  File "/Users/askpatrickw/.pyenv/versions/3.9.0/lib/python3.9/logging/__init__.py", line 1079, in emit
    msg = self.format(record)
  File "/Users/askpatrickw/.pyenv/versions/3.9.0/lib/python3.9/logging/__init__.py", line 923, in format
    return fmt.format(record)
  File "/Users/askpatrickw/.pyenv/versions/3.9.0/lib/python3.9/logging/__init__.py", line 659, in format
    record.message = record.getMessage()
  File "/Users/askpatrickw/.pyenv/versions/3.9.0/lib/python3.9/logging/__init__.py", line 363, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/Users/askpatrickw/.pyenv/versions/cp_base/bin/circup", line 33, in <module>
    sys.exit(load_entry_point('circup==0.2.2', 'console_scripts', 'circup')())
  File "/Users/askpatrickw/.pyenv/versions/3.9.0/envs/cp_base/lib/python3.9/site-packages/click/core.py", line 1025, in __call__
    return self.main(*args, **kwargs)
  File "/Users/askpatrickw/.pyenv/versions/3.9.0/envs/cp_base/lib/python3.9/site-packages/click/core.py", line 955, in main
    rv = self.invoke(ctx)
  File "/Users/askpatrickw/.pyenv/versions/3.9.0/envs/cp_base/lib/python3.9/site-packages/click/core.py", line 1514, in invoke
    super().invoke(ctx)
  File "/Users/askpatrickw/.pyenv/versions/3.9.0/envs/cp_base/lib/python3.9/site-packages/click/core.py", line 1279, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/askpatrickw/.pyenv/versions/3.9.0/envs/cp_base/lib/python3.9/site-packages/click/core.py", line 710, in invoke
    return callback(*args, **kwargs)
  File "/Users/askpatrickw/.pyenv/versions/3.9.0/envs/cp_base/lib/python3.9/site-packages/circup.py", line 592, in main
    logger.info("### Started {}", datetime.now())
Message: '### Started {}'
Arguments: (datetime.datetime(2020, 10, 16, 16, 17, 26, 984865),)
```

I'm am 100% new school which means I really only use (and understand) f-strings.
I converted all the `.format()` and `" {}", var` to f-strings and now it works.

I'll try a few more commands, but I wanted to run this by you first. Are you Ok with that kind of change @dhalbert ?
If so I'll submit a PR. I know I'm basically reformatting and people don't like that sometimes, but it was easier for me to fix it this way. Let me know if that's cool with you.

askpatrickw, 2020-10-28:
When this https://github.com/actions/virtual-environments/issues/1740 is closed then click will update

mattpackwood, 2020-10-29:
@askpatrickw Many thanks!

askpatrickw, 2020-11-05:
PR for Click to test on 3.9 was accepted. https://github.com/pallets/click/pull/1707 but not really needed. 
PRs #40 + #41 = Good to go

PR #40 (2020-11-06): Bump version of click in package dependencies.
* This PR fixes #39 (no known package of `Click` for Python3.9).
* I've allowed the latest version of `Click`(rather than a pinned version) to be installed.
* This is a quick fix and I don't currently have any devices to hand in order to test this change (sorry).
* Before merging please `pip install .` in a Python3.9 virtualenv and try the command out to ensure the bump in version for the package `Click` doesn't break anything.

Of course, given the quick nature of this PR please don't hesitate to discard it.
closed 2020-11-06 by ntoll