adafruit/circup #43: Error upgrading

mattpackwood, 2020-12-03:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.): macOS 11.0.1 (20B29)

- Python version (run `python -version` or `python3 -version`): `Python 3.9.0`

- Error message you are receiving, including any Python exception traces: 

```
    ERROR: Command errored out with exit status 1:
     command: /usr/local/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/s7/typ53r597p33k6rpcwn9lg_40000gn/T/pip-install-3v99wxbx/circup_9246775460bf450a8492b2ae1fdc046c/setup.py'"'"'; __file__='"'"'/private/var/folders/s7/typ53r597p33k6rpcwn9lg_40000gn/T/pip-install-3v99wxbx/circup_9246775460bf450a8492b2ae1fdc046c/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/s7/typ53r597p33k6rpcwn9lg_40000gn/T/pip-pip-egg-info-tedtn3re
         cwd: /private/var/folders/s7/typ53r597p33k6rpcwn9lg_40000gn/T/pip-install-3v99wxbx/circup_9246775460bf450a8492b2ae1fdc046c/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/s7/typ53r597p33k6rpcwn9lg_40000gn/T/pip-install-3v99wxbx/circup_9246775460bf450a8492b2ae1fdc046c/setup.py", line 21, in <module>
        with open(os.path.join(base_dir, "CHANGES.rst"), encoding="utf8") as f:
    FileNotFoundError: [Errno 2] No such file or directory: '/private/var/folders/s7/typ53r597p33k6rpcwn9lg_40000gn/T/pip-install-3v99wxbx/circup_9246775460bf450a8492b2ae1fdc046c/CHANGES.rst'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

```
- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 
`python3 -m pip install --upgrade --user apprise circup `

askpatrickw, 2020-12-04:
I'm not officially supporting this, but I sorta love circup so let me see if I can help.
I tried to reproduce your issue and could not. The only difference in my attempt was I did not use the --user flag. (I use pyenv on MacOS). That _might_ be the issue... but unclear. 

what I ran: `pip install circup==0.1.0` and  `pip install --upgrade circup`. I also tried installing from the repo with a version <0.2.2 and upgrading to the  published version. All worked.

Can you provide the following version information.
`python3 -m pip --version`
`python3 --version`
`circup --version`

I'm also curious if you previously installed from a clone of the repo or only by using pip.

Pip has been undergoing a lot of changes recently so I'm especially intrigued by the versions there.

ntoll, 2020-12-05:
:wave: Hi folks...

The actual error that causes the problem is this:

```
FileNotFoundError: [Errno 2] No such file or directory: '/private/var/folders/s7/typ53r597p33k6rpcwn9lg_40000gn/T/pip-install-3v99wxbx/circup_9246775460bf450a8492b2ae1fdc046c/CHANGES.rst'
```

I don't think this has anything to do with the `--user` flag. :shrug: 

The error basically means that the installer can't find the `CHANGES.rst` file. This **should** be a part of the archive you download from PyPI when you use the `pip` command because the file is explicitly referenced in the `MANIFEST.in` file.

I'm not sure what's going on -- I've just checked out the package that's on PyPI ([this one](https://files.pythonhosted.org/packages/10/a7/58e3512c1e5e37dd609bbf582f6c2153496df8fd88d989b3dea8a5cff4c6/circup-0.2.2.tar.gz) uploaded in August) and can confirm that the `CHANGES.rst` file is actually **IN** the package in the expected location.

I can successfully install `circup` via `pip` (on Linux with Python 3.8). I wonder, as @askpatrickw mentions,

> Pip has been undergoing a lot of changes recently so I'm especially intrigued by the versions there.

Could this be something to do with `pip` versions, Python versions, OS version being a bad/strange combination.

Could you try installing `circup` in a virtualenv please, and let us know what happens..?

Thanks.

mattpackwood, 2020-12-05:
Ok, so here goes...
```
Last login: Thu Dec  3 19:31:08 on console
Matt-Packwoods-iMac:~ $ python3 -m pip --version
pip 20.3.1 from /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pip (python 3.9)
Matt-Packwoods-iMac:~ $ python3 --version
Python 3.9.0
Matt-Packwoods-iMac:~ $ circup --version
CircUp, A CircuitPython module updater. Version 0.2.2
Matt-Packwoods-iMac:~ $ 
```

I installed through pip

```
Matt-Packwoods-iMac:~ $ python3 -m pip uninstall circup                        
Found existing installation: circup 0.2.2
Uninstalling circup-0.2.2:
  Would remove:
    /Users/mattpackwood/Library/Python/3.9/bin/circup
    /Users/mattpackwood/Library/Python/3.9/lib/python/site-packages/circup-0.2.2-py3.9.egg-info
    /Users/mattpackwood/Library/Python/3.9/lib/python/site-packages/circup.py
Proceed (y/n)? y
  Successfully uninstalled circup-0.2.2
```

And things go a little more interesting:
Matt-Packwoods-iMac:~ $ python3 -m pip install --user circup         
```
Collecting circup
  Using cached circup-0.2.2.tar.gz (17 kB)
Requirement already satisfied: semver>=2.8.1 in ./Library/Python/3.9/lib/python/site-packages (from circup) (2.10.2)
Requirement already satisfied: appdirs>=1.4.3 in ./Library/Python/3.9/lib/python/site-packages (from circup) (1.4.4)
Requirement already satisfied: requests>=2.22.0 in ./Library/Python/3.9/lib/python/site-packages (from circup) (2.24.0)
Collecting Click==7.0
  Downloading Click-7.0-py2.py3-none-any.whl (81 kB)
     |████████████████████████████████| 81 kB 2.7 MB/s 
Requirement already satisfied: certifi>=2017.4.17 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from requests>=2.22.0->circup) (2020.6.20)
Requirement already satisfied: chardet<4,>=3.0.2 in ./Library/Python/3.9/lib/python/site-packages (from requests>=2.22.0->circup) (3.0.4)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in ./Library/Python/3.9/lib/python/site-packages (from requests>=2.22.0->circup) (1.25.10)
Requirement already satisfied: idna<3,>=2.5 in ./Library/Python/3.9/lib/python/site-packages (from requests>=2.22.0->circup) (2.10)
Using legacy 'setup.py install' for circup, since package 'wheel' is not installed.
Installing collected packages: Click, circup
  Attempting uninstall: Click
    Found existing installation: click 7.1.2
    Uninstalling click-7.1.2:
      Successfully uninstalled click-7.1.2
    Running setup.py install for circup ... done
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
black 20.8b1 requires click>=7.1.2, but you have click 7.0 which is incompatible.
Successfully installed Click-7.0 circup-0.2.2
Matt-Packwoods-iMac:~ $ 
```

ntoll, 2020-12-05:
@mattpackwood so,

```
Successfully installed Click-7.0 circup-0.2.2
```

Suggests it worked..? If so, can we close this..?

Thanks.

mattpackwood, 2020-12-05:
I am going to close it, but...

When I update only circup, no issue, if I update with other things, I get errors...  I think the suggestion above about pip having issues is probably correct, I will dig further but I think this is a "close"

closed 2020-12-05 by mattpackwood

ntoll, 2020-12-05:
@mattpackwood aha. That makes sense. Thank you for diligently following this through and sorry it's not something we can fix. I guess you could use `circup` as an example case for a bug in `pip`..? See https://github.com/pypa/pip