adafruit/circup #48: --version does not show the correct version

askpatrickw, 2020-12-21:
Since the fix for #19  the --version outputs the place holder.

```circup --version
CircUp, A CircuitPython module updater. Version 0.0.0-auto.0
```

It should output the correct version. (note: Pip list does show the correct version).

askpatrickw, 2020-12-21:
Some lunch thoughts:

- module.__version__ is apparently not a standard thing. It was trying to be but the PEP was deferred. 
  - https://stackoverflow.com/questions/20180543/how-to-check-version-of-python-modules
  - PEP 396 https://www.python.org/dev/peps/pep-0396/
- `from importlib.metadata import version` and then `version('module')` does work for circup. Because it is using the CPython standard library and because it matches pip list.. it seems like a _"better/more correct"_ method. But it is not 100% true in all places and is => Python 3.8
- I could also see just removing --version since pip list gives you this info.

makermelissa, 2020-12-21:
Yeah, this will likely need to be updated with Github Actions to automatically replace it like the other libraries, but it will probably need some digging in.

askpatrickw, 2020-12-22:
@makermelissa  and @kattni  not sure who should review... but the fix was simple. Click does this for us, we were just not letting it do its thing.

PR #49 (2020-12-22): Don't pass version and Click will figure it out
This fixes #48 

Click has builtin cross python support to figure this out and we are using click's version_option, but we were passing __version__. If we don't do that, Click figures it out for us correctly.
closed 2020-12-22 by makermelissa