adafruit/circup #50: Install dependencies

askpatrickw, 2020-12-22:
When you are installing one package it may also have other dependencies called out, install those as well.

Not sure how this will work yet, but is the next thing that messes me up in my own workflow where I'm having to get one package then another package.

Not included: uninstall dependencies. Pip doesn't even do that yet.

askpatrickw, 2021-01-06:
I was thinking that there would some way to deduce that from the library, but I see what you're pointing out.
CP libraries aren't PyPI packages so they don't follow the [package format](https://python-packaging.readthedocs.io/en/latest/dependencies.html). 

Circup is just downloading and copying files. Maybe we could include a library_name_requirements.txt file in the distributions... or maybe a longer-term support for PyPi style packages can be investigated (not required for now).

Just thoughts... ;-)

makermelissa, 2021-01-06:
I'm not sure what the advantage of having library_name_requirements.txt would be since we already have the url for the repo and can just download the requirements.txt at the root of that.

askpatrickw, 2021-01-14:
@makermelissa, that's a great point!
Circup won't work offline anyway so its not like grabbing a text file is that big a deal. We would need to figure out the CPython vs. CircuitPython  dependencies since Blinka, pyserial, etc. are in the requirements for many (all?) the packages.

Related: I'm taking notes on  if\how pip could be used for CPython library installs on devices over in this gist [Package Management for CircuitPython](https://gist.github.com/askpatrickw/e69eacd2c60c8e27f255d976ab5e8820)

makermelissa, 2021-01-14:
I see. Yeah, maybe some kind of ignore list. There's only like 6 modules/packages that might be listed that are not CircuitPython related.

PR #69 (2021-01-27): Install libraries' dependencies
- fixes #50
- Alphabetized the imports
- Alphabetize the functions (Sorry, I know this makes the diff big, but it was hard to find stuff)
- New functions: clean_library_name(), get_dependencies(), libraries_from_requirements()
- Refactored: install()

Fixed along the way (my iterative notes)
 - [x] Right now this works, but it makes circup unbearably slow because its getting dependencies (hitting github) for _every_ library in the bundle. I need to move things around to speed this up so I only call get_dependencies() during the actual installation.
     - Cleane [... cut]
closed 2021-01-27 by makermelissa