adafruit/circup #55: Circup does not support multiple CP devices attached at the same time

cpforbes, 2020-12-31:
When I have 2 circuit python devices attached it will only report on the first one.

A --path option that will override the built in device path detection would fix this.

I will submit a PR that adds a --path option.

askpatrickw, 2020-12-31:
I honestly had never plugged in more than one board before you mentioned this.
But that's a great idea for an an optional parameter.

Neradoc, 2021-01-04:
I rename my circuitpython boards with names like CLUE, CIRCUITBLUE, FEATHERS2... so I added the option to give it a volume name (not case sensitive), which I think is more friendly than a path. It wouldn't work with two drives with the same name, but it works with a full path since it uses a `volume.endswith` call.

Except that circup on posixes uses `mount` to find the volume and splits on spaces, and macOS renames a second CIRCUITPY board to `/Volumes/CIRCUITPY 1` so it will not catch it. So I switched to `psutil.disk_partitions()` instead (which adds a dependency). On windows it only uses the volume's name, I'm not sure how to do better. Recognize an a name like "E:" ?

These work on my mac setup:
`circup --name clue list`
`circup --name circuitpy list`
`circup --name "CIRCUITPY 1" list`
`circup --name /Volumes/CIRCUITPY\ 1 list` (what you get by drag-and-dropping into the terminal)

makermelissa, 2021-01-05:
Fixed via #60

closed 2021-01-05 by makermelissa