adafruit/circup #56: Make check runs pyflakes and pycodestyle, CI runs Pylint

askpatrickw, 2020-12-31:
The make check is not running the same linting as the CI process.

They should be the same. Will ask on Discord about CP preferences.

askpatrickw, 2021-01-01:
@kattni  and @tannewt  I'm not sure who should chime in on this. But if there is some preferred list of checks for the cpython code let me know. My default would be to have the checks\tests in the repo match the github action. tx!

ladyada, 2021-01-02:
hihi folks will be back in a few days!

kattni, 2021-01-04:
@askpatrickw Hey Patrick! I agree it should be the same. We run Pylint and Black. Sphinx builds the docs, though I'm not certain that's required for this.

makermelissa, 2021-01-04:
I agree with @Kattni. This one is kind of the odd one out and running it on the 2 she mentioned should be good.

PR #63 (2021-01-06): Black and pylint Suport in dev environment
Fixes #56 by removing pyflakes and pycodestyle from the makefile and make.py and replacing with black and pylint along with updating  setup.py and requirements.txt for same.

Tested locally
closed 2021-01-06 by makermelissa