adafruit/circup #59: crash on macos due to extended attribute files (._*)

Neradoc, 2021-01-04:
When running circup on macOS 10.15.7, with python 3.8, I encounter the following error:
```
❯ circup update --all
Found device at /Volumes/CIRCUITPY, running CircuitPython 6.0.1.
Found 1 module[s] needing update.
Something went wrong, [Errno 2] No such file or directory: '.___init__.mpy' (check the logs)
```

From the log file:
```
12/29/2020 14:54:12 ERROR: [Errno 2] No such file or directory: '.___init__.mpy'
Traceback (most recent call last):
  File "/Users/spyro/Library/Python/3.8/lib/python/site-packages/circup.py", line 695, in update
    module.update()
  File "/Users/spyro/Library/Python/3.8/lib/python/site-packages/circup.py", line 175, in update
    shutil.rmtree(self.path)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/shutil.py", line 715, in rmtree
    _rmtree_safe_fd(fd, path, onerror)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/shutil.py", line 672, in _rmtree_safe_fd
    onerror(os.unlink, fullname, sys.exc_info())
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/shutil.py", line 670, in _rmtree_safe_fd
    os.unlink(entry.name, dir_fd=topfd)
FileNotFoundError: [Errno 2] No such file or directory: '.___init__.mpy'
```

Which seems to be a general problem with `shutil.rmtree`:

```
>>> import shutil
>>> shutil.rmtree("/Volumes/CIRCUITPY/lib/adafruit_display_text/")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/shutil.py", line 715, in rmtree
    _rmtree_safe_fd(fd, path, onerror)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/shutil.py", line 672, in _rmtree_safe_fd
    onerror(os.unlink, fullname, sys.exc_info())
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/shutil.py", line 670, in _rmtree_safe_fd
    os.unlink(entry.name, dir_fd=topfd)
FileNotFoundError: [Errno 2] No such file or directory: '.___init__.mpy'
```

The extended attribute files are not created by circup, but is there if you copied the modules to the board manually in the past. It would be better if circup did not choke on them.

I fixed it for me by using: `shutil.rmtree(temp_dir, ignore_errors=True)` everywhere rmtree is used, but I don't know if that's the appropriate way to do that, or what other genuine errors it might hide, or if I'm the only one with the problem, or if it's a matter of python version or...

makermelissa, 2021-01-05:
I looked into this a bit further and it's not actually crashing for me (at least on the latest version), but it is outputting warnings.

makermelissa, 2021-01-05:
I looked into this further and it appears to be a bug in shutil. I like the solution you provided and couldn't find a better one, except I will only add it to the problematic one for now.

PR #61 (2021-01-06): Ignore error caused by bug in shutil.rmtree
Fixes #59.
closed 2021-01-06 by jepler