adafruit/circup #68: Allow some commands to run without a board connected

askpatrickw, 2021-01-21:
Now that you can use `circup show <query>` to find things, it would be nice not to have to have a board attached to do so.

dgriswo, 2021-08-18:
The CircuitPython version is read from boot_out.txt.  The version determines the bundle to query against.

What are thoughts on adding a BOARDLESS_COMMANDS list and testing ctx.invoked_subcommand against that list. (Currently only show would be present.)  The CPY_VERSION would be set to get_latest_release_from_url (currently 6.3.0, even with 7.0.0-alpha in flight)

If a board is detected, use the version in boot_out.txt

askpatrickw, 2021-08-18:
I think when I opened this we may only have supported one bundle, but regardless, I like both of your ideas.
Would you also be able to specify the version or only use the default (6.x now)?

How about a PR and we try it?

dgriswo, 2021-08-19:
Submitted https://github.com/adafruit/circup/pull/121 for just the boardless show against the latest.

I'm thinking a cleaner approach might be another subcommand.  Maybe `circup show-version 7.3.0`  This would simplify the show command and allow querying a different version if you did have a board attached.

Neradoc, 2021-08-19:
Modules are not versioned by Circuitpython version (currently), the CPY version is only used to chose the MPY format to install, or test if there is a new core update, it really doesn't matter otherwise. The show command would do the exact same thing, no matter what version you give, it always uses the `py` version of the bundle as reference.

The install command requires a version, I would like for `install`:
- the `--py` option also removes the requirement for a valid CPY version (even for a boot_out.txt file)
-  `-6` and `-7` or `--cp 6.3.0` or something like that to similarly force the use of a MPY format of the other.

dgriswo, 2021-08-19:
I see.  After allowing 'show' to run without a board, I got a ValueError on CPY_VERSION and set it to latest to fix that.  Took another peek at the code and see that the error was coming from the code block checking if the board version is older than latest.  If that check is only performed when a device_path is specified/detected, there is no reason to set CPY_VERSION for the show command.  I'll update the PR.

I can take a shot at the `install` changes.  I think `--cp 6.3.0` would survive version updates better than `-6` or `-7`.  Is there an API that lists all valid version numbers that the user input could be validated against?

Thanks for your patience.  I'm relatively new to python and am just trying to give back to the community where I can.

Neradoc, 2021-08-19:
> Is there an API that lists all valid version numbers that the user input could be validated against?

I don't know that there is a great source for actual version numbers, maybe get the release tags from the github API ? That might be slow (and rate limited - but that's unlikely to be an issue) though it could be cached.

> Thanks for your patience. I'm relatively new to python and am just trying to give back to the community where I can.

Your involvement is welcome, I'm happy to see more activity on circup.
It does motivate me to finish a couple of PRs I've been pushing back for weeks. ;)

bablokb, 2022-04-30:
I actually solved the problem for me. I keep my sources per project below `projectname/src` (and not on the device itself). Then I copy an existing `boot_out.txt` to the `src` directory and then run e.g.

    circup --path src list

This allows me to update the libs on my PC. During development, I then use rsync to copy the contents of `projectname/src` to my test-devices. Of course I need to keep the `boot_out.txt` in sync with the real CP version, so  a commandline-option would be much appreciated (but even then I would have to remember the correct version).

askpatrickw, 2022-12-25:
I didn't test every command, but I think this is fixed now. Should we close this?

```bash
circup show adafruit_requests
Downloading latest version for adafruit/Adafruit_CircuitPython_Bundle.

  [####################################]  100%
  [####################################]  100%
  [####################################]  100%

OK

adafruit_requests
1 shown of 396 packages.

circup --version
CircUp, A CircuitPython module updater. Version 1.1.4

circup install adafruit_requests
Could not find a connected CircuitPython device.

circup freeze
Could not find a connected CircuitPython device.
```

hyx0329, 2024-03-31:
Hi, currently the `install` command won't work without a `boot_out.txt` even if a desired cpy version passed through the command line. Is that intentional?

The backend(`DiskBackend`) uses `self.get_circuitpython_version()`, the one reads `boot_out.txt`, to get current board/cpy version. User-configured version info is not passed to this backend object when it's instantiated.

My use case is packaging all code and essential libraries to a tarball for redistribution.

bablokb, 2024-03-31:
Have you seen my workaround posted above? https://github.com/adafruit/circup/issues/68#issuecomment-1113983786

Your tar-command must then explicitly exclude the `boot_out.txt` when creating the tarball, but that is no problem.

hyx0329, 2024-03-31:
@bablokb Yeah I've read it. Thanks. It's just a strange thought in my mind for I have to generate/keep one `boot_out.txt` in my project repository.

bablokb, 2024-03-31:
True, I have IT in all of my projects and in all of my `.gitignore`-files.