adafruit/circup #70: DUNDER_ASSIGN_RE method for determining __repo__ in extract_metadata() does not work on multi-line values

askpatrickw, 2021-01-24:
The way extract_metadata() walks a .py file _assumes_ all dunder values are on the same line as the variable name

This does not work if the repo url is long enough that black will reformat it and wrap with parenthesis.

So far, I've only seen this with 1 library:

```python
__version__ = "2.0.4"
__repo__ = (
    "https://github.com/adafruit/Adafruit_CircuitPython_BLE_BerryMed_Pulse_Oximeter.git"
)
```

I wonder if we can't import __repo__ and __version__ or the whole library and get all the dunder globals... but I've not thought about it much yet.

makermelissa, 2021-01-25:
I think it has to do with longer strings. Importing everything might be a slow process. Perhaps we can update to account for the new format.

askpatrickw, 2021-01-25:
I should have written the description a little differently...
The reason the multiline value fails is file is split on "\n" an then walked line by line...
https://github.com/adafruit/circup/blob/master/circup.py#L330

Hmm... we could read in the file, remove all new lines and then regex out a value... Feels like a hack though. ;-)

makermelissa, 2021-01-25:
You can do multiline regex. I'm just checked and it's doing regex currently, but likely isn't taking the parenthesis into account and may need to be adjusted to work on multiple lines. I did some multiline regex when working on this library: https://github.com/adafruit/Adafruit_Python_Shell/blob/master/adafruit_shell.py, but I can't remember the specifics.

askpatrickw, 2021-01-26:
This is also fixed in my PR.
I took a simpler route. Since we were already reading in the whole file, I could look for all dunder matches, loop through them and add them to result without the use of exec() and without looping over the whole file line by line.

commit 2ed81dd5 (2021-01-27): fix #70
closed 2021-01-27 by makermelissa