adafruit/circup #71: Handle default branch name properly

askpatrickw, 2021-01-26:
The name of the default branch can change. Currently many use `master`. Many people are switching to `main`.
Ideally we can handle it being _anything_.

This affects the url used to retrieve the requirements.txt file in `libraries_from_requirements()`.

makermelissa, 2021-01-26:
Oh, nice catch.

askpatrickw, 2021-01-26:
@makermelissa  Yeah, I saw on discord that cater was going to change the branch on the nunchuk lib repo. So my known issue is now a problem. 🐛

makermelissa, 2021-01-26:
Easy way would be to attempt to use main and then fall back to master, but I'm poking around to see if there's a cleaner way.

askpatrickw, 2021-01-26:
With a get on the repo URL, we could find the URL to the requirements.txt in the reponse.
`r = requests.get("https://github.com/adafruit/circup")`

`<a class="js-navigation-open link-gray-dark" title="requirements.txt" href="/adafruit/circup/blob/master/requirements.txt">requirements.txt</a>`

and get the path to the file with regex

`href="(.+)\>requirements\.txt<\/a>`

makermelissa, 2021-01-26:
Great solution

askpatrickw, 2021-01-26:
I'm not fan of regex... so I kept looking...
https://docs.github.com/en/rest/reference/repos#get-a-repository

```python
url = "https://api.github.com/repos/adafruit/circup"
d = requests.get(url)
d.json()['default_branch']
'master'
```

makermelissa, 2021-01-26:
That seems less hacky. :)

askpatrickw, 2021-01-27:
In fixing this, I discovered two things:

1. several repos had invalid __repo__ values.
2. I wasn't checking if I got a requirments.txt back.

Not the end of the world, but I'll do a PR to fix this after the other PR is approved since that change so big.

PR #72 (2021-02-05): retrieve default branch via regex
- Fixes #71 
- Moved skipping non-mcu libraries to logging, output is cleaner now
- Moved some logic into its own function `get_requirements()` which makes `get_dependencies()` a little cleaner too

I pretty quickly hit rate limiting using the GitHub API, so I am using a regex approach similar to what's described in the issue.

https://docs.github.com/en/rest/overview/resources-in-the-rest-api#rate-limiting
"For unauthenticated requests, the rate limit allows for up to 60 requests per hour. Unauthenticated requests are associated with the originating IP addres [... cut]
closed 2021-02-05 by makermelissa