adafruit/circup #74: Mu Editor Integration

makermelissa, 2021-01-30:
This would be useful to add to Mu (the original author of this library was also one of the lead developers for mu). I don't believe mu supports plugins, so it would have to be added as a feature.

ntoll, 2021-01-30:
Would love to help with this. Perhaps you could describe how you think this might work in terms of the UI..? We'd just add such a feature to the Adafruit mode..! :-)

makermelissa, 2021-02-05:
I'm not sure. Maybe a couple buttons or a dialog?

askpatrickw, 2021-02-06:
I'm not a Mu user, but I could see a few things, mimicking the CPI:

1. Import requirements.txt which would call circup and install libraries 
2. Freeze : show currently installed libraries 
4. Install single library
5. Upgrade device libraries

closed 2024-12-20 by dhalbert