adafruit/circup #79: Bad version strings make circup crash

Neradoc, 2021-02-07:
```
❯ circup --path /Volumes/UMFS2/ list
Found device at /Volumes/UMFS2/, running CircuitPython 6.1.0.
A newer version of CircuitPython (6.2.0-beta.0) is available.
Traceback (most recent call last):
  File "circup.py", line 1054, in <module>
    main()
  File "/Library/Python/3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Python/3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Library/Python/3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Python/3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Library/Python/3.8/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "circup.py", line 881, in list
    modules = [m.row for m in find_modules(ctx.obj["DEVICE_PATH"]) if m.outofdate]
  File "circup.py", line 881, in <listcomp>
    modules = [m.row for m in find_modules(ctx.obj["DEVICE_PATH"]) if m.outofdate]
  File "circup.py", line 193, in row
    major_update = str(self.major_update)
  File "circup.py", line 174, in major_update
    == VersionInfo.parse(self.bundle_version).major
  File "/Users/spyro/Library/Python/3.8/lib/python/site-packages/semver.py", line 726, in parse
    raise ValueError("%s is not valid SemVer string" % version)
ValueError: v3.3.0 is not valid SemVer string
```
The culprint is `adafruit_si7021` whose bundle version string is `v3.3.0`.
Also looking at all the version strings pulled from the latest bundle, `adafruit_rplidar` has `0.0.1-auto.0` and makes circup crash too.
Every other modules follows the 1.2.3 format.

I guess fixing the tag in Adafruit_CircuitPython_SI7021 is the simplest way to fix that, but we may want to make circup more resilient to that kind of issue.
https://github.com/adafruit/Adafruit_CircuitPython_SI7021/tags

I wonder why adafruit_rplidar has a wrong version (latest release is 1.1.4)
https://github.com/adafruit/Adafruit_CircuitPython_RPLIDAR/tags

askpatrickw, 2021-02-07:
Thanks @Neradoc for reporting!

GitHub suggests adding the "v" in the release page, but I think its been a common convention for the CP Community to not add it. 

> Tagging suggestions
> It’s common practice to prefix your version names with the letter v. Some good tag names might be v1.0 or v2.3.4.
> 

We shouldn't crash for that. The easy fix is to strip "v", but I'm debating if we should be validating the __Version__ values in some rigorous way or just having a convention and doing what you suggested, open bugs in repos that don't follow it. 

Any thoughts?

askpatrickw, 2021-02-07:
I believe adafruit_rplidar has a bug and the auto-version isn't setup right. 
Circup is finding all those. ;-) 

I will look at it more closely Sunday or Monday night.

makermelissa, 2021-02-07:
When a release is published, our convention is to not have the v. Occasionally people publish it that way and it doesn't get caught. So maybe use regex to validate, attempt to strip the v and revalidate and fall back to giving a warning.

tannewt, 2021-02-09:
Note, we don't use the `v` prefix because MicroPython does.

askpatrickw, 2021-02-09:
Thanks @tannewt ! Its good to understand the reasoning.

askpatrickw, 2021-02-09:
I released Adafruit_CircuitPython_SI7021 with the corrected tag
https://github.com/adafruit/Adafruit_CircuitPython_SI7021/releases/tag/3.3.1

Also submitted a PR for RPLIDAR to correct dunder version

PR #80 (2021-02-14): Accept version strings with v
Adds a "clean_version" function that removes the v in front of the version number so semver won't choke on it. There is a related idea that does a little more in the semver documentation:
https://python-semver.readthedocs.io/en/2.13.0/usage.html#dealing-with-invalid-versions

Also changes expecting a _`ValueError` in addition to a `TypeError`_ in `Module.major_update()`, so that bad version strings are caught. This fixes crashing circup. Remaining bad version numbers are assumed to be major updates.

This PR fixes #79 in regards to the adafruit_si7021 version. But [... cut]
closed 2021-02-14 by askpatrickw