adafruit/circup #81: Use requirement.txt in the bundle downloads

askpatrickw, 2021-02-11:
@jepler  is adding the requirements file (if one exists) to the bundle download.
https://github.com/adafruit/circuitpython-build-tools/pull/60

The requirements are within the bundle like so:
`Adafruit_CircuitPython_Bundle-6.x-mpy-20210209/requirements/adafruit_seesaw/requirements.txt`

`BUNDLE_RELEASE/requirements/LIBRARY_NAME/requirements.txt`

To remove the extra network traffic, use this instead.

askpatrickw, 2021-02-13:
I have this blocked out... waiting for new bundles to be released with the requirements included to officially test. 
1.7.0 released 13 Feb ~ Noon so sometime after 14 Feb

jepler, 2021-02-13:
I see you made a release of the updated circuitpython-build-tools, :tada: so now we just need to have some updated libs to release..

askpatrickw, 2021-02-14:
Already in progress ;-)

jepler, 2021-02-14:
whee -- https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20210214

now we just need a community bundle update

jepler, 2021-02-14:
I've just published a release of https://github.com/jepler/Jepler_CircuitPython_udecimal/releases/tag/1.0.2 so tomorrow's adabot run should give us an updated community bundle.

askpatrickw, 2021-02-14:
Community bundle is an open issue, but one I'd like to add #82

PR #83 (2021-02-18): Use requirements from the bundle 
fixes #81 

This uses requirements from the bundle.
Note that it only looks for requirements in the py bundle path. I don't know of a reason we should look in the MPY bundle instead of the py bundle for this information.
closed 2021-02-18 by askpatrickw