adafruit/circup #82: Support the Community Bundle

askpatrickw, 2021-02-11:
Circup should also support the community bundle.

askpatrickw, 2021-03-09:
This could use an in the weeds conversation, some thoughts:

1. Can we just have one bundle download for circup?
2. If there are two individual downloads, should circup search through them as if they are one? OR should you have to specify --community or some such flag?

tannewt, 2021-03-09:
1. I think we'll want multiple. I envision separate bundles for other company sponsored drivers too such as sparkfun and pimoroni.
2. I'd do all by default and only require specifics when two bundles provide the same thing.

jposada202020, 2021-04-03:
This would be a great addition :)

arturo182, 2021-05-15:
This seems to be the place to ask: is there a plan to allow non-adafruit libraries in circup? I would love to be able to tell users of my HW to just do a `circup install xxx` and it will install my libraries and pull in the adafruit requirements :)

ladyada, 2021-05-15:
we'd be happy to take a PR for adding support of the community bundle (or any other number of bundles) - we dont have it specifically planned out.

askpatrickw, 2021-05-15:
100% everyone would love to see this added.  Pls do PR if you can.  If not someday I will do it.

arturo182, 2021-05-15:
Since I haven't looked at the code at all yet, any suggestions on how you see this implemented are welcome :)

tannewt, 2021-05-18:
> Since I haven't looked at the code at all yet, any suggestions on how you see this implemented are welcome :)

I always imagined other companies would have bundles of their own and tools like `circup` would know about them all. The community bundle is a special case for one-off third-party libraries.

arturo182, 2021-05-18:
Yes, I would assume I just maintain my own bundle with my libraries.

I meant more on the implementation side, right now the Adafruit bundle is very much baked in (https://github.com/adafruit/circup/blob/master/circup.py#L455). I'd assume we move this info out into a json file so it's easy to make PRs with new bundles, and users can also easily add new bundles locally if they want to. This also means that we need to keep tag info for each bundle so we know when they go out of data, etc.

askpatrickw, 2021-05-18:
The Community and the Adafruit bundle could be added internally with a CLI parameter for a bundle URL (or maybe a bundle_config.json) which has a URL to an unknown bundle. My thought was that circup would fetch all bundles, search all as one set of libraries. 

> Note: That would require libraries to not have collision in naming across bundles or some sort of prioritization hierarchy.

I also think it makes sense to add 3rd party bundles by adding them to circup repo maintained bundle_config.json. So you could have "public" bundles in the bundle_config.json and private bundles added via the CLI or modifying the local bundle_config.json.

The format of a bundle is sort of documented by the fact that the only bundles are created by the same scripts, as long as we expect new bundles to use those scripts that's adequate for now. But a bundle verification test would be a nice add to that process\repo.

todbot, 2021-07-12:
Looks like this was added via https://github.com/adafruit/circup/pull/110  Thank you @Neradoc!

askpatrickw, 2021-08-03:
22 days late, but closing this because its already in and working. ;-)

closed 2021-08-03 by askpatrickw