adafruit/circup #90: Missing dependency

rpavlik, 2021-03-14:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.): windows 10 64-bit

- Python version (run `python -version` or `python3 -version`):  Python 3.9.0

- Error message you are receiving, including any Python exception traces: 

```
code.py output:
Traceback (most recent call last):
  File "code.py", line 4, in <module>
  File "adafruit_emc2101.py", line 30, in <module>
ImportError: no module named 'adafruit_register'
```

- List the steps to reproduce the problem below (if possible attach code or commands
  to run):

On a pretty clean install of circuitpython, run `circup install adafruit_emc2101`

It says there are no dependencies, but when you run it, it complains of missing adafruit_register.

ladyada, 2021-03-14:
hmm its in the dependency list
https://github.com/adafruit/Adafruit_CircuitPython_EMC2101/blob/main/requirements.txt

@makermelissa @askpatrickw either of ya'll wanna see whats up?

Neradoc, 2021-03-14:
I believe circuit-build-tool needs a release to deploy this:
https://github.com/adafruit/circuitpython-build-tools/pull/62

askpatrickw, 2021-03-14:
@Neradoc is correct, in that that PR would be the fix. If you look in circup's cached bundle there is no requirements file for adafruit_emc2101.

```ll ~/Library/Application\ Support/circup/adafruit_circuitpython_bundle_py/adafruit-circuitpython-bundle-py-20210314/requirements/adafruit_emc*
zsh: no matches found: /Users/patrickwalters/Library/Application Support/circup/adafruit_circuitpython_bundle_py/adafruit-circuitpython-bundle-py-20210314/requirements/adafruit_emc*
```

That issue was fixed 9 days ago, but last release was 29 days ago... I'll test @Neradoc's PR and do a release.

askpatrickw, 2021-03-14:
Actually... I can't really test that... so I just did a code review and will now do a release. ;-)

ladyada, 2021-03-14:
@askpatrickw thank you 🙏 let me know if you need any permission bits to do the release

askpatrickw, 2021-03-14:
No worries, @ladyada. Happy to help!

The fix for this is in 1.7.1 of circuitpython-build-tools and all the bundles will be updated at the stroke of midnight.

closed 2021-03-14 by askpatrickw

askpatrickw, 2021-03-16:
Just an FYI, as of the 16 March 2021 Bundle release this is now fixed.

langestefan, 2021-05-25:
@askpatrickw 
> Just an FYI, as of the 16 March 2021 Bundle release this is now fixed.

Hi Patrick. I am still facing exactly this issue after installing adafruit_mcp9808. I check circuitpythonbuildtools and it's at version 1.8.5. Any idea what I can do?

` python3 mcp9808.py                  Traceback (most recent call last):
  File "mcp9808.py", line 3, in <module>
    import adafruit_mcp9808
  File "/usr/local/lib/python3.7/dist-packages/adafruit_mcp9808.py", line 41, in <module>
    from adafruit_register.i2c_bits import RWBits
ModuleNotFoundError: No module named 'adafruit_register'`

lesamouraipourpre, 2021-05-25:
> @askpatrickw
>
>
> Hi Patrick. I am still facing exactly this issue after installing adafruit_mcp9808. I check circuitpythonbuildtools and it's at version 1.8.5. Any idea what I can do?
>
> ` python3 [...]

There is a pending PR for the adafruit_mcp9808 library that will fix this. It will hopefully get merged and release in the next 24-48 hours.
https://github.com/adafruit/Adafruit_CircuitPython_MCP9808/issues/30

For now, if you run the following it should fix your problem:
`circup install adafruit-circuitpython-register`

langestefan, 2021-05-25:
@lesamouraipourpre Thank you. That worked :)

tony359, 2021-06-10:
Hello all

I am having the same issue

```
from adafruit_register.i2c_bits import RWBits
ModuleNotFoundError: No module named 'adafruit_register'
```

When I run 
`circup install adafruit-circuitpython-register` I get a `bash: circup: command not found`

I did install circup using 
`pip3 install --user circup`

Can anybody help me please? Thank you!

jposada202020, 2021-06-10:
@tony359 it seems that you do not have circup installed, follow the installations instructions here https://github.com/adafruit/circup
according to your system

tony359, 2021-06-10:
Thank you. I did install it before, I tried again and it told me it was already installed... but now it works. (maybe because before the default Python version was 2.7, now it's 3.7)

Ok, now I get this

`Could not find a connected Adafruit device.`

which I am not sure is relevant to this thread to be honest. The sensor is definitely connected and it was working before on Stretch (yes, I did a clean install).

jposada202020, 2021-06-10:
Well that is an improvement, maybe post your question in the Adafruit Discord channel. Normally Github is used for libraries development.

tony359, 2021-06-10:
I shall do - thanks for your help.

One thing though - the previous message said that the dependency was supposed to be fixed 2 weeks ago. Just out of curiosity, I guess this is not the case yet?

Thanks

Neradoc, 2021-06-10:
This issue was about `circup` not installing all dependencies.
If you can't even run `circup`, that has nothing to do with this issue.

tony359, 2021-06-10:
When I run my code, it still says that adafruit-register is not available. Circup was to be used as a workaround because the adafruit-register dependency was missing - in my understanding, apologies if I am not 100% accurate on this, I am not a developer.

I am looking into why circup throws up that error message.

By my question stand: is the adafruit-register dependency still missing from the module which I have installed? 

Thank you!

jposada202020, 2021-06-10:
@tony359 what library are you trying to use? MCP9808 EMC2101 or any other? please contact us in Adafruit discord channel: #help-with-circuitpython I do now want to `spam` anymore this issue. Thanks :)