adafruit/circup #91: Detect and output script dependencies

jwcooper, 2021-03-18:
A recent feature request for learn.adafruit.com was to automatically detect and bundle any library dependencies for a CircuitPython script as part of the download.

An example would be for the script found here:
https://learn.adafruit.com/adafruit-ultimate-gps/circuitpython-python-uart-usage#example-parsing-code-3003973-6

We'd want to know that `adafruit_gps` is utilized, but also any adafruit libraries that `adafruit_gps` imports.

I'm not sure if it's feasible, but I'm hoping something like `circup dependencies project.py` would be something that would be possible as a new feature.

The only alternative I've found that can get me close to this functionality is the [findimports](https://pypi.org/project/findimports/) project.

This would also likely be dependent on https://github.com/adafruit/circup/issues/68 as we wouldn't want to run this on a connected board.

ladyada, 2021-03-18:
we're thinkin of turning this into a 'updated once a night' json file generated by gh actions. melissa will update ya as we get there :)

askpatrickw, 2021-03-18:
Similar to PIP, you can have a project specific requirements.txt would that solve this and not require a new command?

> Note: Circup would also pull in dependencies for anything in the requirements.txt.

```
circup install --help
Found device at /Volumes/CIRCUITPY, running CircuitPython 6.2.0-beta.3-105-g526348066-dirty.
Usage: circup install [OPTIONS] [MODULES]...

  Install a named module(s) onto the device. Multiple modules can be
  installed at once by providing more than one module name, each separated
  by a space. Option -r allows specifying a text file to install all modules
  listed in the text file.

  TODO: Ensure there's enough space on the device, work out the version of
  CircuitPytho on the device in order to copy the appropriate .mpy versions
  too. ;-)

Options:
  --py
  -r, --requirement TEXT
  --help                  Show this message and exit.
  ```

askpatrickw, 2021-03-18:
Oh wait, I get it , You want to make a project download... So you want to release a .zip for a learn guide with all the things in it.... interesting.

makermelissa, 2021-03-18:
The idea is really a multi-part one. First I need to make an API with similar functionality as CircUp that does some of the module cataloging and sorting with the JSON file. I plan to use JavaScript to do this and have it live on GitHub. It will be useful for plugin development to make it easier, but also to be able to request a URL with certain libraries, and have a nice zip bundle returned with each of the requested libraries plus their dependencies. First step is to have adabot generate and upload a JSON file nightly that resides on GitHub somewhere (probably on circuitpython.org).

makermelissa, 2021-03-18:
A related issue to this is: https://github.com/adafruit/circuitpython-org/issues/491

jwcooper, 2021-03-18:
This all sounds good to me. Thanks for the link to that other issue as well, hadn't seen that somehow (even though I'm mentioned multiple times :/)

makermelissa, 2021-03-25:
Currently json is included as an attachment on Bundle Release Assets, but we're looking at moving a copy of the assets to Amazon S3.

askpatrickw, 2021-08-03:
@makermelissa  didn't you have custom dependency downloads for learn guides working? I remember you working on it. Was wondering if we can close this.

makermelissa, 2022-02-11:
Yep, I think we can close it.

closed 2022-02-11 by makermelissa