adafruit/circup #93: Check CircuitPython version

lesamouraipourpre, 2021-03-23:
Feature request

Add support for checking the installed version of CircuitPython and advising if there is a newer version (or beta version) available. 
While CircUp won't be able to install the update, it could provide a link to the download location or download it if requested.

lesamouraipourpre, 2021-03-24:
After a proper look at the code, CircUp already has this feature for normal releases.

It would be nice to be notified of newer beta versions, but after reading github documentation the only redirecting url seems to be github.../releases/latest which links to the current non-beta release.

Without parsing the github..circuitpython/releases (or maybe github..circuitpython/tags) page, I can't see an easy way to extract the information. 

I'll leave this open in case someone can figure an easy way to get the latest beta/rc release version. Otherwise, I'll close it in a week or so.

Neradoc, 2021-03-24:
The list of releases if available from: `https://api.github.com/repos/adafruit/circuitpython/releases`
A non stable relase is one where `"-" in release['tag_name']` (that's the test circuitpython itself does to identify un/stable releases in the download pages of circuitpython.org)
```py
import json
import requests
response = requests.get("https://api.github.com/repos/adafruit/circuitpython/releases")
data = response.json()
betas = [
    release['tag_name']
    for release in data
    if "-" in release['tag_name']
]
print(betas[0])
```

lesamouraipourpre, 2021-03-24:
```
betas = [ release['tag_name'] for release in data if "-" in release['tag_name'] ]
```
could be improved to
```
betas = [ release['tag_name'] for release in data if release['prerelease'] ]
```

We could also do something with git command line like
`git ls-remote --tags --refs https://github.com/adafruit/circuitpython`
This currently downloads ~11k compared to ~100k for the webpage. It also gives all tags but they would need to be version parsed and sorted.

askpatrickw, 2021-03-24:
Thanks @lesamouraipourpre for going back and verifying that indeed circup does have this already.

It might be worth adding the downloads link into the existing new version notification.
https://circuitpython.org/downloads

https://github.com/adafruit/circup/blob/master/circup.py#L819

As for discovering and encouraging beta usage, for the general population of CP folks (beginners) I wouldn't push for that. It's also not how python packaging\pip works. You pip install from the releases and you go to the repo\website for non-released versions.  I think we're trying to mostly mimic that behavior here (although we use "list" in non pip-like fashion).

lesamouraipourpre, 2021-03-25:
We could do something like the following pseudo-code which for most CircUp users would do nothing different. But for users who currently have a prerelease version installed it will run extra version checks.
```
if current_version < latest_stable_version:
    display update message

if current_version is a prerelease version:
    get latest release versions

    if current_version < latest_prerelease_version_with_same_major:
        display update message

    if current_version_major < maximum_prerelease_major:
        display information about a new version being developed
```

For example (with made up version numbers):

Example A - stable user:
Current Version : 6.0.2
There is a newer version of CircuitPython available: 6.1.0

Example B - testing user:
Current Version : 6.0.1-beta.2
There is a newer version of CircuitPython available: 6.1.0
There is a newer prerelease version of CircuitPython available: 6.2.0-rc.4
There is a future version of CircuitPython being developed: 7.0.0-alpha.2

lesamouraipourpre, 2021-03-25:
As an aside, version 6 is currently hard-coded into the script:
```
def get_bundle(tag):
    urls = {
        "py": (
            "https://github.com/adafruit/Adafruit_CircuitPython_Bundle"
            "/releases/download"
            "/{tag}/adafruit-circuitpython-bundle-py-{tag}.zip".format(tag=tag)
        ),
        "6mpy": (
            "https://github.com/adafruit/Adafruit_CircuitPython_Bundle/"
            "releases/download"
            "/{tag}/adafruit-circuitpython-bundle-6.x-mpy-{tag}.zip".format(tag=tag)
        ),
    }
```

Is this an intentional behaviour that is planned for in the transition to CP7? Or should I raise an issue about it?

askpatrickw, 2021-03-25:
It used to have both 5 and 6 listed so I think you're correct it is intentional and where one could add 7 and the community bundle. 

If there's a better way I know everyone appreciates PRs. :)

makermelissa, 2021-03-25:
Oh, by the way, there is now a JSON file attached to the Bundle releases that includes all the libraries, versions, and dependencies (as well as paths in the bundle itself). I'm working on an API that will allow you to query all this information as well to facilitate plugin development and make things easier that makes use of the JSON file.

lesamouraipourpre, 2021-04-06:
I'm closing this as the upcoming work to be done on moving to JSON dependency management should hopefully cover most cases.

closed 2021-04-06 by lesamouraipourpre