adafruit/circup #96: Requirements that are not published on PyPi

askpatrickw, 2021-03-26:
I was adding adafruit_logging as a dependency to adafruit_azureiot and ran into this issue where Adafruit-CircuitPython-Logging isn't published on pypi, so the build failed.

I don't have any thoughts about that yet.. but I wanted to capture the issue for now.
@makermelissa  you probably are thinking about this as well.

See Related:
https://github.com/adafruit/Adafruit_CircuitPython_AzureIoT/pull/29

tannewt, 2021-03-29:
It shouldn't be on pypi because CPython has its own version. How does circup use pypi?

makermelissa, 2021-03-29:
Oh, hi @askpatrickw. I missed this. Yes, I have a solution. As part of the bundle builds, we are now publishing a JSON file as both part of the release assets and uploaded to Amazon S3. You can download the assets (bundle zips and json files) off S3 at https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bundles/adafruit/ and https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bundles/community/

askpatrickw, 2021-03-30:
The way circup determines dependencies is by opening the libraries(s) requirements.txt from within the bundle which was [added a while back](https://github.com/adafruit/circuitpython-build-tools/pull/60). It de-dupes and installs the complete chain of dependencies.

The issue I found in adding adafruit_logging to adafruit_azureiot's requirements.txt was that the build failed because adafruit_logging is not published on PyPi.

I've not thought too hard about this yet, but it seems like there is no way to express a dependency for a library that is CP only and not also Blinka supported in a manner that is commonly practiced in a common Python mechanism.

makermelissa, 2021-03-31:
Right, I understand. The json files I am building are done at the time the bundles themselves are created. It makes use of the requirements files for determining the dependencies.

I have just created a PR to update the JSON file with the main change being libraries are listed by their bundle module/package name, but I am also providing a PyPI name as well. There are now dependencies and external_dependencies fields depending on whether the dependency was found inside the bundle or externally (mostly for the community bundle libraries depending on libraries in the Adafruit bundle). For the external dependency, I am using the PyPI name.

Anyway, the point is, you can use the JSON to get the info you need regardless of whether it is published on PyPI or not as long as it's part of a bundle.

makermelissa, 2021-03-31:
PR is merged, you can either get the updated JSON files off S3 or wait until tomorrow for the updated release assets.

askpatrickw, 2021-03-31:
Not sure I'm following... 
If the JSON is generated from the requirements.txt and if adding a library  to requirements.txt which is not published on PyPI breaks the build for a library. I don't think the existence of the JSON changes that. 

As for should circup use the JSON. Is there some reason to do that? I'm not sure I see that one works differently than the other. I'm not against the JSON at all, I just see it being for a different purpose (learning guides).

makermelissa, 2021-03-31:
Ok, @askpatrickw, I'm following along now.

What I've seen other repositories do it to create a separate requirements.txt file. So we could do something like circuitpython_requirements.txt that is only used for indexing purposes. Another option would be to have my json file builder actually scan imports and get a more accurate list of requirements, though I think that is a potentially buggier way of doing it and may take quite a while.

askpatrickw, 2021-03-31:
I was writing a novel in here... I'll show up on Monday with an in the weeds topic instead.

askpatrickw, 2021-04-07:
This was discussed during [In The Weeds on 5 April](https://github.com/adafruit/adafruit-circuitpython-weekly-meeting/blob/master/2021/2021-04-05.md#4553-in-the-weeds).

## Conclusions from that discussion

- [ ] CircuitPython only requirements which are in addition to those in requirements.txt will be documented in a circuitpython-requirements.json -- @MakerMelissa will decide the filename\format.
    - [ ] This should be added\documented\mentioned (?) in the Cookiecutter repo
- [ ] Yes. Circup will move to the JSON files.
    - [ ] The JSON files with dependencies will need to be added to the bundle downloads.
    - [ ] When circup is  updated to use the JSON, the requirements.txt in the bundles can be removed.
> Yes. It is planned that in the future where there are 3-rd party bundles they’d be built the same way.

makermelissa, 2021-04-07:
As discussed in the meeting, we decided including a circuitpython-requirements.json in some libraries. I think the format should just be something simple like:
```
{
	"requirements": [
		"adafruit_ssd1305",
		"adafruit_requests",
		"adafruit_logger"
	]
}
```
We may need to add some kind of json validation to the GitHub actions for the libraries.

askpatrickw, 2022-12-25:
@makermelissa and @Neradoc (since you've made a lot of fixes) I'm curious if you think this issue is still relevant? 

If you're not sure, let me know and I'll do some testing in the new year.