adafruit/circup #98: Fails when board reports version 7

jepler, 2021-04-07:
```
$ circup --version
CircUp, A CircuitPython module updater. Version 0.9.4
$ circup install adafruit_hid
Found device at /media/jepler/CIRCUITPY, running CircuitPython 7.0.0-alpha.1-3-gfd17c8f24-dirty.
Searching for dependencies for: ['adafruit_hid']
Ready to install: ['adafruit_hid']

Traceback (most recent call last):
  File "/home/jepler/.local/bin/circup", line 8, in <module>
    sys.exit(main())
  File "/home/jepler/.local/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/jepler/.local/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/jepler/.local/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/jepler/.local/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/jepler/.local/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/jepler/.local/lib/python3.7/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/jepler/.local/lib/python3.7/site-packages/circup.py", line 942, in install
    install_module(ctx.obj["DEVICE_PATH"], library, py, mod_names)
  File "/home/jepler/.local/lib/python3.7/site-packages/circup.py", line 749, in install_module
    raise IOError("Cannot find compiled version of module.")
OSError: Cannot find compiled version of module.
```

makermelissa, 2021-04-07:
I think if the bundle isn't available for the found version, it should fall back to the latest one. I think the circuitpython vscode extension may have the same issue.

askpatrickw, 2021-04-07:
What triggers the existence of a CP7 bundle ?

makermelissa, 2021-04-07:
I think this will probably fix itself after tonight's bundle is released.

askpatrickw, 2021-04-08:
Fingers crossed 🤞

askpatrickw, 2021-04-08:
Oh yeah... we do need to add 7mpy, see linked PR.

tannewt, 2021-04-08:
https://github.com/adafruit/circuitpython-build-tools/pull/68 Should enable the 7.x build. I do want to caution that the mpy version for 7.x will change if we do merge in newer micropython.

todbot, 2021-05-17:
If anyone else comes here having problems getting circup to install 7.x mpys, I found that if I deleted the circup data dir and tried again, it worked.  For example, upgrade to 0.9.5 circup and try to install `neopixel` on QT Py:

```
$ circup --version
CircUp, A CircuitPython module updater. Version 0.9.5
$ circup install neopixel
Found device at /Volumes/CIRCUITPY, running CircuitPython 7.0.0-alpha.2-586-g0ac107b73.
Searching for dependencies for: ['neopixel']
Ready to install: ['adafruit_pypixelbuf', 'neopixel']

Traceback (most recent call last):
  File "/usr/local/bin/circup", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/circup.py", line 947, in install
    install_module(ctx.obj["DEVICE_PATH"], library, py, mod_names)
  File "/usr/local/lib/python3.9/site-packages/circup.py", line 754, in install_module
    raise IOError("Cannot find compiled version of module.")
OSError: Cannot find compiled version of module.
```

So then delete circup data dir:

```
$ circup_dir=`python -c 'import appdirs; print(appdirs.user_data_dir(appname="circup", appauthor="adafruit"))'`
$ echo $circup_dir
/Users/tod/Library/Application Support/circup
$ rm -rf $circup_dir
```

And then try again:

```
$ circup install neopixel
Found device at /Volumes/CIRCUITPY, running CircuitPython 7.0.0-alpha.2-586-g0ac107b73.
Downloading latest version information.

OK

Searching for dependencies for: ['neopixel']
Ready to install: ['adafruit_pypixelbuf', 'neopixel']

Installed 'adafruit_pypixelbuf'.
Installed 'neopixel'.
```