adafruit/circup #100: Handle modules that are moved from a single file to a package

Neradoc, 2021-04-17:
```
Update 'adafruit_emc2101'? [y/N]: y
Something went wrong, [Errno 2] No such file or directory: '/Users/spyro/Library/Application Support/circup/adafruit_circuitpython_bundle_7mpy/adafruit-circuitpython-bundle-7.x-mpy-20210417/lib/adafruit_emc2101.mpy' (check the logs)
```
That's because the module changed to a `adafruit_emc2101` directory.

askpatrickw, 2021-04-20:
@Neradoc , Does this happen when a library was previously published as a single file, but now it is a folder based library?

Neradoc, 2021-04-20:
Yes, sorry I wasn't very clear:
- the library was originally a `adafruit_emc2101.mpy` file, and is installed as such on the board.
- it was since [reworked into multiple files](https://github.com/adafruit/Adafruit_CircuitPython_EMC2101/pull/6) in a `adafruit_emc2101` folder.
- when running `circup update` the error above appears, and the mpy file is deleted.

The exception is:
```py
[Errno 2] No such file or directory: '/Users/neradoc/Library/Application Support/circup/adafruit_circuitpython_bundle_6mpy/adafruit-circuitpython-bundle-6.x-mpy-20210420/lib/adafruit_emc2101.mpy'
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/circup.py", line 1053, in update
    module.update()
  File "/usr/local/lib/python3.9/site-packages/circup.py", line 196, in update
    shutil.copyfile(self.bundle_path, self.path)
  File "/usr/local/Cellar/python@3.9/3.9.4/Frameworks/Python.framework/Versions/3.9/lib/python3.9/shutil.py", line 264, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/Users/neradoc/Library/Application Support/circup/adafruit_circuitpython_bundle_6mpy/adafruit-circuitpython-bundle-6.x-mpy-20210420/lib/adafruit_emc2101.mpy'
```
Because the code assumes it's always the same type both when finding the bundle module and copying the files.
https://github.com/adafruit/circup/blob/0d9219baa28c394d712c79c0b28495b14128264e/circup.py#L125-L128
https://github.com/adafruit/circup/blob/0d9219baa28c394d712c79c0b28495b14128264e/circup.py#L189-L196

FoamyGuy, 2021-11-20:
I ran across this issue today as well with a very old single file version of `adafruit_pyportal` not being able to successfully update the modern version which is now a directory.

b-blake, 2024-06-10:
Hi,

It does it with **adafruit_ov5640.mpy** also.  May 31, 2024 was my experiance.