adafruit/circup #104: Missing dependency

LenReinhart, 2021-04-23:
Using the RP2040 Feather with the APDS9960 the following error came up after loading the color example:
File "code.py", line 7, in <module>
  File "adafruit_apds9960/apds9960.py", line 16, in <module>
ImportError: no module named 'adafruit_register'

Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **INSERT PLATFORM/OPERATING
  SYSTEM HERE**

- Python version (run `python -version` or `python3 -version`):  **INSERT PYTHON
  VERSION HERE**

- Error message you are receiving, including any Python exception traces:  **INSERT
  ERROR MESAGE/EXCEPTION TRACES HERE***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**

askpatrickw, 2021-04-23:
This is because the  Adafruit_CircuitPython_APDS9960 does not have adafruit_register. In its requirements.txt
https://github.com/adafruit/Adafruit_CircuitPython_APDS9960/blob/master/requirements.txt

Can you open an issue in that repo and do a PR to fix?

askpatrickw, 2021-04-23:
hmm... looking at this more closely. I see it in requirements.txt.

askpatrickw, 2021-04-23:
My test:

```
circup install adafruit_apds9960

Found device at /Volumes/CIRCUITPY, running CircuitPython 6.2.0-beta.3-104-g7c610bdd2-dirty.
A newer version of CircuitPython (6.2.0) is available.
Searching for dependencies for: ['adafruit_apds9960']
Ready to install: ['adafruit_apds9960', 'adafruit_bus_device', 'adafruit_register']

Installed 'adafruit_apds9960'.
'adafruit_bus_device' is already installed.
Installed 'adafruit_register'.
```

@LenReinhart  Can you check your version of circup for me by running circup --version. It should be 0.9.4

LenReinhart, 2021-04-23:
Where do I do this? I tried from >>> connected to the RP2040 and got:

 

Press any key to enter the REPL. Use CTRL-D to reload.

 

Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit Feather RP2040 with rp2040

>>> circup --version

Traceback (most recent call last):

File "<stdin>", line 1, in <module>

NameError: name 'circup' is not defined

askpatrickw, 2021-04-23:
@LenReinhart  [Circup is a command-line too](https://github.com/adafruit/circup/blob/master/README.rst)l run from your desktop\laptop to install CircuitPython libraries onto your device. 

Are you sure you're using Circup?

LenReinhart, 2021-04-23:
I thought Circup was adafruit shorthand for Circuit Python, I looked it up on the Adafruit site and have installed it. My device now says Adafruit Register is installed and I will proceed from here. Assume problem solved. Thanks, without your involvement I would have not known about Circup.

askpatrickw, 2021-04-23:
No worries @LenReinhart. Happy to help.  There is a great community on Discord and a #help-with-circuitpython channel you might find useful.  https://adafru.it/discord

closed 2021-04-23 by askpatrickw