adafruit/circup #112: Should Circup move to the CircuitPython Organization?

askpatrickw, 2021-06-02:
@tannewt  asked about moving Circup to the CircuitPython organization on Discord and I thought we could track\discuss in an  Issue better. 

@lesamouraipourpre, @Neradoc and @joedevivo do you have any concerns about making such a move?

It makes sense to me. Feels very much like a tool the community should (does) support.
Please "@" anyone else who should be informed\consulted. ty!

joedevivo, 2021-06-02:
Honestly, I didn't know there was a CircuitPython organization. What's the plan for that?

lesamouraipourpre, 2021-06-02:
> Honestly, I didn't know there was a CircuitPython organization. What's the plan for that?

http://github.com/circuitpython

I _think_ the idea is to host repositories that are related to CircuitPython but not related to Adafruit hardware. Only 5 libraries have been moved over so far and it's a work in progress to get the principles and infrastructure (like circup and cookiecutter) correct.

I believe @FoamyGuy and @jposada202020 are taking the lead with the moves so far as they have been working on graphical libraries.

> @tannewt asked about moving Circup to the CircuitPython organization on ~Discord~**Github** and I thought we could track\discuss in an Issue better.
> 
> It makes sense to me. Feels very much like a tool the community should (does) support.

With respect to the `circup` move, I'm in favour of it. I believe most users who use circup will acquire it and update it from PyPI, so it should be possible to transfer ownership of the library such that users don't even notice. While it would need the secrets (or suitable maintainer secrets) for PyPI and readthedocs, IIRC @tannewt said that was either already setup or could be setup easily.

A logical follow-on of moving circup to `http://github.com/circuitpython` would be to move the CircuitPython Community Bundle.

joedevivo, 2021-06-02:
I'm not actually using circup anymore for the vscode extension. It was easier to store the state of the Adafruit bundle in memory doing some inspection of the downloaded bundle. That said, I don't have any issue with moving circup, and I'm open to the idea of moving [vscode-circuitpython](https://github.com/joedevivo/vscode-circuitpython) into the CircuitPython org if that seems like the right move. 

Since that's kind of moving off topic, I don't mind moving that conversation somewhere else, I'm just not sure where that somewhere else would be.

Neradoc, 2021-06-03:
I don't really have an opinion on that, how will it impact reviews and getting updates in ?

askpatrickw, 2021-06-03:
The first step is to get all the open PRs merged.... then we can ask an Adafruit Repo Admin to do the move.

FoamyGuy, 2021-06-03:
I believe the idea is that libraries supported primarily by the community rather than Adafruit directly will live inside of the circuitpython org.

Indeed the first ones we are doing are some of the graphical helper libraries that were originally in the DisplayIO_Layout library.

There were some changes in cookie cutter to support the desired naming for these projects moving forward, and I think all of those are merged in now so we are ready to get back into creating new repos for those to live in. I'll get back to doing some more of those this week.

As for circup, I do think it would make sense to move to the circuitpython org. But I am not super familiar with it's internals, I don't know if there is anything that would need to be changed / fixed in order to support the repo moving places (though github does seem to do some redirecting so possibly not).

askpatrickw, 2021-06-03:
Circup  pulls from the bundle releases but it doesn't read its own repo (that I recall)

tannewt, 2021-06-03:
> The first step is to get all the open PRs merged.... then we can ask an Adafruit Repo Admin to do the move.

I *think* PRs will come with this repo if I just transfer ownership. The one thing we may need to do is add a pypi maintainer since we have a different account for the CP org. (The credentials are already set.)

askpatrickw, 2021-06-03:
> I _think_ PRs will come with this repo if I just transfer ownership. The one thing we may need to do is add a pypi maintainer since we have a different account for the CP org. (The credentials are [...]

You're right... I was thinking of something Kattni said about it being easier to manage the branch name change to main when there are no PRs.

jwcooper, 2021-09-16:
learn.adafruit.com is pulling from the [bundle sources json](https://github.com/adafruit/circup/blob/main/circup/config/bundle_config.json) now. If this moves orgs, we'll need a small update to learn as well.