adafruit/circup #113: Provide more help when there is a new CP Release

askpatrickw, 2021-06-27:
Circup tells you "A newer version of CircuitPython (6.3.0) is available."

Circup should provide a link to the board specific download page.
The boot_out.txt may provide enough clues to get there, or at enough to send the user to the Downloads page with a query parameter

```
cat /Volumes/CIRCUITPY/boot_out.txt
Adafruit CircuitPython 6.2.0-beta.4-93-g0105b30a0 on 2021-03-25; Adafruit Feather M4 Express with samd51j19
```
Unforturnately the The boot_out.txt doesn't provide a workable query

https://circuitpython.org/downloads?q=Adafruit Feather M4 Express with samd51j19  
^^ doesn't work

https://circuitpython.org/downloads?q=Feather+M4+Express  
^^ that does

Some way to link a board a learn guide about a board, a board's download page would be ideal.
Either a board_id or common string in the  CP Build the CP.org page, etc...

lesamouraipourpre, 2021-06-27:
I think the easiest fix to this would be a change in `circuitpython` that changes the `boot_out.txt`
```diff
diff --git a/py/makeversionhdr.py b/py/makeversionhdr.py
index a89eb35e1..c40fd1459 100644
--- a/py/makeversionhdr.py
+++ b/py/makeversionhdr.py
@@ -107,7 +107,7 @@ def make_version_header(filename):
 #define MICROPY_VERSION_MINOR (%s)
 #define MICROPY_VERSION_MICRO (%s)
 #define MICROPY_VERSION_STRING "%s"
-#define MICROPY_FULL_VERSION_INFO ("Adafruit CircuitPython " MICROPY_GIT_TAG " on " MICROPY_BUILD_DATE "; " MICROPY_HW_BOARD_NAME " with " MICROPY_HW_MCU_NAME)
+#define MICROPY_FULL_VERSION_INFO ("Adafruit CircuitPython " MICROPY_GIT_TAG " on " MICROPY_BUILD_DATE "; " MICROPY_HW_BOARD_NAME " (" CIRCUITPY_BOARD_ID ") with " MICROPY_HW_MCU_NAME)
 """ % (
         git_tag,
         git_hash,
```

On a QT-Py this produces:
```
Adafruit CircuitPython 7.0.0-alpha.3-95-g2548f865f-dirty on 2021-06-27; Adafruit QT Py M0 (qtpy_m0) with samd21e18
```

The text in the brackets is the BOARD name used to build the board.
We can then use the link: `https://circuitpython.org/board/<BOARD>` , eg.
https://circuitpython.org/board/qtpy_m0

I don't know what effect this would have on any other parts of CircuitPython so it would need more knowledgable discussion.

askpatrickw, 2021-06-27:
@kattni and @tannewt  it would be good to get your input here.

Summary: Can we add "board id" to the boot_out.txt so we can use it to reliably link to the board pages in CP.org?

Concerns: 

1. There would be something in the CP.org repo and the CP repo that would need to be kept in-sync. Although it appears as if the CP.org URL matches the folder name in https://github.com/adafruit/circuitpython/tree/main/ports/esp32s2/boards 
2. Do we know of any tools or process that rely on the boot_out.txt format?

tannewt, 2021-06-28:
I like the idea of making the board id available from CircuitPython. I did that somewhere once.... I'd like to see it available to Python code as well, not just boot_out.txt.

Do you do any discovery through serial or only from boot_out.txt? I think it's ok to evolve it.

Neradoc, 2021-06-28:
Circup just reads the `boot_out.txt` file. (Which incidentally makes it easy to simulate different versions with local directories containing a boot_out file and --path for tests).

Also as far as I know the board id is identical on circuitpython.org and boot_out.txt, except for the alias boards that use the exact same binary as the original, so the boot_out contains the original's id, but it would still be the correct file so it's not a big deal.
https://github.com/adafruit/circuitpython/blob/b265a924d119eb40e35353246818158a0c8ae267/docs/shared_bindings_matrix.py#L35-L44

lesamouraipourpre, 2021-06-29:
> I like the idea of making the board id available from CircuitPython. I did that somewhere once.... I'd like to see it available to Python code as well, not just boot_out.txt.

It could be added to `board` as `board.BOARD_NAME`
or
It could be added to `os.uname()` probably as `nodename`. (to quote `man 2 uname` - `the field _nodename_ is meaningless`)
Currently all ports return the same pair of values  for [0]`sysname` and [1]`nodename`, eg:
```python
>>> os.uname()
(sysname='samd51', nodename='samd51', release='7.0.0',
 version='7.0.0-alpha.2-646-g7df5d74d9 on 2021-05-22', machine='Adafruit PyPortal with samd51j20')
```

https://github.com/adafruit/circuitpython/blob/826e25989c99ca698b39843ce4f32203ca6de7ff/ports/atmel-samd/common-hal/os/__init__.c#L41-L51

askpatrickw, 2021-08-03:
@tannewt What do you think of @lesamouraipourpre's suggestion for adding board id to `board` and\or `os.uname()['nodename']` ?

tannewt, 2021-08-04:
`board.ID` would be my preference I think. That might risk conflicting with an ID pin though. `board.NAME` would be ok but name implies to me that it is human readable. I'd drop the `BOARD_` prefix because the module name does it already.

askpatrickw, 2021-08-06:
OK, so thanks to @Neradoc's discussion on Discord looks like they have the path forward on this nailed down. 

For Circup, board_id (the folder name of board in ports..) can be added to boot_out.txt and if we also want to add that to board.ID, that can be handled in a different issue in CP.

askpatrickw, 2021-09-06:
This is fixed by #124

closed 2021-09-06 by askpatrickw