adafruit/circup #114: Move to MAIN branch

askpatrickw, 2021-06-27:
@kattni  there are no open PRs. Can you please have Circup setup to use MAIN as the default branch.

kattni, 2021-06-28:
@askpatrickw Hello! I have moved this repo to `main`. I did NOT, however, delete the `master` branch yet. I have protected it so no one can push to it, but I didn't want to remove it until you're certain that nothing else is using it. Please look into it and let me know when you're ready for that branch to be deleted. (For example, the VS Code CircuitPython Plugin appears to use circup, and I have no idea the extent to which that is the case.)

I'm going to leave this issue open. Please ping me directly when you're certain that the `master` branch can be deleted.

askpatrickw, 2021-06-28:
@joedevivo  and @saschakiefer GH is saying you both has Circup as a dependency. When we delete the master branch and make main the default, will this break you?

askpatrickw, 2021-06-30:
@kattni  I say go head and delete MASTER. The other "depends" on repo is just someone's personal samples. Nothing "production".

kattni, 2021-06-30:
Done!

closed 2021-06-30 by kattni