adafruit/circup #122: Several Libraries have dependencies which fail to install

askpatrickw, 2021-08-20:
None or some of these may be a circup issue. Just capturing this while I look into it.

- [ ]     {'adafruit_pyoa': 'adafruit_display_button'},
- [ ]     {'arrowline': 'vectorio'},
- [ ]     {'jepler_udecimal': 'astroid'},

FoamyGuy, 2021-08-20:
I'm not sure about the others, but `vectorio` is a built-in module it doesn't need to be installed with circup, it's included in the core of builds that can use it.

Neradoc, 2021-08-20:
`adafruit_button` is one of those Adafruit libraries that have a non regular name and should likely be added to the `not_standard_names` list like `adafruit_sd` and the others.
```py
        "adafruit_display_button": "adafruit_button",
```

askpatrickw, 2021-08-20:
> I'm not sure about the others, but `vectorio` is a built-in module it doesn't need to be installed with circup, it's included in the core of builds that can use it.

@FoamyGuy  But if you're using the library with Blinka, you will need to `pip install vectorio`.

FoamyGuy, 2021-08-20:
> > I'm not sure about the others, but `vectorio` is a built-in module it doesn't need to be installed with circup, it's included in the core of builds that can use it.
> 
> @FoamyGuy But if you're using the library with Blinka, you will need to `pip install vectorio`.

@askpatrickw  As far as I know there is no CPython / Blinka implementation of `vectorio`. There is this PyPi entry, but I believe it's unrelated to circuitpython: https://pypi.org/project/vectorio/

If it did exist I think it would be inside of https://github.com/adafruit/Adafruit_Blinka_Displayio rather than it's own library, but am not certain of that.

askpatrickw, 2021-08-24:
Thanks @FoamyGuy, I just assumed they were the same. We should have Arrowline change. I'll follow-up.

askpatrickw, 2021-09-07:
All the affected libraries are updated

closed 2021-09-07 by askpatrickw