adafruit/circup #127: 3rd Party Bundle Support

askpatrickw, 2021-09-08:
Today (2021-09-08) Circup can download and install libraries from three bundles:

- Adafruit CircuitPython Bundle
- Community Bundle
- CircuitPython.org Bundle

This issue\discussion is about how to support 3rd party bundles which follow the CP library bundling format. This conversation originally started in a PR to move the references to the bundles from within the code to an JSON configuration file. (https://github.com/adafruit/circup/pull/126)  

Proposals at that time included:

## User Supplied Config.json

In addition to the config.json released with circup,  look for a config file in [AppDirs](https://pypi.org/project/appdirs/) so that you can look for the config json file first in the user's configuration directory and then to the default list shipped in the package. If a config.json existed in AppDirs, those bundles would be merged with the bundles released with Circup.

## Circup Bundle Management Commands

Add Bundle Management Commands to Circup for _Add_, _Remove_, _Reorder_, display _Info_ for bundles. This would update the configuration JSON.

> I think that summarizes things.. I'll add my own comments\thoughts seperately

askpatrickw, 2021-09-08:
I actually don't have any big thoughts about this right now. I think the file in AppDirs is a good plumbing step and then the commands could be added separately.

Are there any third party library bundles today?
It might also be a useful mechanism for testing.

Neradoc, 2021-09-08:
I've been testing https://github.com/Neradoc/Circuitpython_Keyboard_Layouts locally with an implementation of bundle commands, I'll look into reworking it to use bundle_config.json.

jepler, 2022-09-12:
Did this get implemented or is there something left to do?