adafruit/circup #134: There was a problem downloading the bundle. Please try again in a moment.

ajs256, 2021-12-19:
- Platform/operating system: macos 12.1

- Python version (run `python -version` or `python3 -version`):  3.9.9

- Error message you are receiving, including any Python exception traces:  
```
ajs256@greeble ~ $ circup install adafruit_macropad
Found device at /Volumes/CIRCUITPY, running CircuitPython 7.0.0.
Downloading latest version for adafruit/Adafruit_CircuitPython_Bundle.

  [####################################]  100%          
There was a problem downloading the bundle. Please try again in a moment.
```

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): Attempt to download any library

FoamyGuy, 2021-12-19:
This should be resolved with #132. Make sure that you have circup versions `1.0.1` https://github.com/adafruit/circup/releases/tag/1.0.1 it was released just the other day so you may need to pip update if you haven't yet.

ajs256, 2021-12-20:
That worked, thanks!

closed 2021-12-20 by ajs256