adafruit/circup #137: uable to install Circup 1.0.2

dsssssssss9, 2021-12-24:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Ubuntu 30.05

- Python version (run `python -version` or `python3 -version`):  3.8.10

- Error message you are receiving, including any Python exception traces:  pip3 install --user  circup
/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
/usr/lib/python3/dist-packages/secretstorage/util.py:19: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
Requirement already satisfied: circup in /usr/local/lib/python3.8/dist-packages (1.0.1)
Requirement already satisfied: appdirs>=1.4.3 in /usr/lib/python3/dist-packages (from circup) (1.4.3)
Requirement already satisfied: findimports>=2.1.0 in ./.local/lib/python3.8/site-packages (from circup) (2.1.0)
Requirement already satisfied: semver~=2.13 in ./.local/lib/python3.8/site-packages (from circup) (2.13.0)
Requirement already satisfied: requests>=2.22.0 in /usr/lib/python3/dist-packages (from circup) (2.22.0)
Requirement already satisfied: Click>=8.0 in ./.local/lib/python3.8/site-packages (from circup) (8.0.3)

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): pip3 --user  install circup

dhalbert, 2021-12-24:
Try `pip3 install --upgrade circup` or `pip3 uninstall circup; pip3 install circup`. I just did the former myself and it worked.

Note that `--user` is now the default on the usual Ubuntu 20.04 `pip3`.

dsssssssss9, 2021-12-24:
hi ...
super quick replay sir

I tried BOTH ways and got the following each time.....

pip3 install circup
/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
/usr/lib/python3/dist-packages/secretstorage/util.py:19: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: circup in /usr/local/lib/python3.8/dist-packages (1.0.1)
Requirement already satisfied: semver~=2.13 in ./.local/lib/python3.8/site-packages (from circup) (2.13.0)
Requirement already satisfied: appdirs>=1.4.3 in /usr/lib/python3/dist-packages (from circup) (1.4.3)
Requirement already satisfied: Click>=8.0 in ./.local/lib/python3.8/site-packages (from circup) (8.0.3)
Requirement already satisfied: findimports>=2.1.0 in ./.local/lib/python3.8/site-packages (from circup) (2.1.0)
Requirement already satisfied: requests>=2.22.0 in /usr/lib/python3/dist-packages (from circup) (2.22.0)
jkd@jkd-mininuc:~$ circup
bash: /home/jkd/.local/bin/circup: No such file or directory

dhalbert, 2021-12-24:
This is fishy: `Requirement already satisfied: circup in /usr/local/lib/python3.8/dist-packages (1.0.1)`. Should not be `/usr/local/lib` for a `--user` install.
It looks like it might have been installed earlier with `sudo pip3`. So try an `sudo pip3 uninstall circup`.

dsssssssss9, 2021-12-24:
Hi

please do feel free to  put this on hold until after the holidays - you have family & friends who deserve your holiday cheer much much more than i need your technical wizardry!!

I did as you suggested 

sudo pip3 uninstall circup
followed by 
pip3 install circup

please see attached text file for screen outpout & errors
[circup problem 1.txt](https://github.com/adafruit/circup/files/7775080/circup.problem.1.txt)

dhalbert, 2021-12-24:
You are much closer, now the issue is an actual bug, that the `update_checker` dependency is not present or not working.

Don't worry about me; I am not working on a holiday for me.

dhalbert, 2021-12-24:
To fix this temporarily, do `pip3 install update_checker`, which is a dependency I forgot.

dsssssssss9, 2021-12-24:
a small xmas present from the ever so helpful ( & let's face it even more ever so patient!!! ) Mr Halbert!!!

circup --version
CircUp, A CircuitPython module updater. Version 1.0.2

dhalbert, 2021-12-24:
I made a new 1.0.3 release, which fixes this. `pip3 install --upgrade circup` will get it. Thanks for your testing today.

closed 2021-12-24 by dhalbert

dsssssssss9, 2021-12-24:
circup --version
CircUp, A CircuitPython module updater. Version 1.0.3

this didn't work
pip3 install --upgrade circup

uninstll followed by install did!!!

many many thanks ....