adafruit/circup #141: Suppress warning about circuitpython_typing

dhalbert, 2022-02-18:
Now that `circuitpython_typing is a separate library, but is not in the bundle (because it's typing definitions only), we are seeing a warning:

```py
$ circup install adafruit_bus_device
Found device at /media/halbert/CIRCUITPY, running CircuitPython 7.2.0-alpha.1-276-g8d4333203.
Searching for dependencies for: ['adafruit_bus_device']
WARNING:
	adafruit_typing is not a known CircuitPython library.
Ready to install: ['adafruit_bus_device']
```
Suppress this warning.

askpatrickw, 2022-02-18:
I think the fix is to add `adafruit_typing` to the list of libraries not intended to be installed on a board.

https://github.com/adafruit/circup/blob/main/circup/__init__.py#L49

dhalbert, 2022-02-18:
The other bug here is that it's not `adafruit_typing`, it's `circuitpython_typing`, and the pypi name is `adafruit-circuitpython-typing`. I'll fix the appropriate libraries.

PR #142 (2022-02-18): special-case circuitpython_typing
- Fixes #141.

Add `circuitpython_typing` to the `NOT_MCU_LIBRARIES` list. Reorder the library canonicalization flow to do the `NOT_MCU_LIBRARIES` check earlier, to avoid mangling the incoming name to soon. Improve variables names in that flow.

Note that `circup` now checks for new versions, so users should get notified this is available when it is released.
closed 2022-02-18 by dhalbert