adafruit/circup #143: ModuleNotFoundError: No module named 'pkg_resources'

CharString, 2022-03-11:
We have a runtime dependency on setuptools:

https://github.com/adafruit/circup/blob/86ff01c5fb33d130ffb8d9ed3ed144cedb220c03/circup/__init__.py#L23

but it's not named in the `install_requires`

https://github.com/adafruit/circup/blob/86ff01c5fb33d130ffb8d9ed3ed144cedb220c03/setup.py#L25-L34

Alternatively we could migrate to `importlib.resources` (new in Python 3.7, but a backport for 3.6 exists)
https://docs.python.org/3/library/importlib.html#module-importlib.resources

CharString, 2022-03-11:
PS This is low priority since it isn't triggered if you use the recommended way of installing. I just bumped into it because I start projects using [pdm](https://pypi.org/project/pdm/):

```console
$ pdm add circup
[…]
$ pdm run circup --help
Traceback (most recent call last):
  File "/home/chars/Projects/spukhaft/__pypackages__/3.7/bin/circup", line 5, in <module>
    from circup import main
  File "/home/chars/Projects/spukhaft/__pypackages__/3.7/lib/circup/__init__.py", line 23, in <module>
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'
```

askpatrickw, 2022-03-11:
@CharString, I'm not sure what PDM is, but please feel free to submit a PR and we're all happy to have a look at what the fix would be.

putz612, 2023-10-14:
Just hit this one on a fresh install of Python 3.12 on Windows

```
PS C:\Users\user> circup list
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\jsievert\AppData\Local\Programs\Python\Python312\Scripts\circup.exe\__main__.py", line 4, in <module>
  File "C:\Users\jsievert\AppData\Local\Programs\Python\Python312\Lib\site-packages\circup\__init__.py", line 23, in <module>
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'
```

Running pip install setuptools fixes this issue

```
PS C:\Users\user> pip3 install setuptools
Collecting setuptools
  Obtaining dependency information for setuptools from https://files.pythonhosted.org/packages/bb/26/7945080113158354380a12ce26873dd6c1ebd88d47f5bc24e2c5bb38c16a/setuptools-68.2.2-py3-none-any.whl.metadata
  Using cached setuptools-68.2.2-py3-none-any.whl.metadata (6.3 kB)
Using cached setuptools-68.2.2-py3-none-any.whl (807 kB)
Installing collected packages: setuptools
Successfully installed setuptools-68.2.2

PS C:\Users\user> circup list
Found device at D:\, running CircuitPython 8.2.6.
Downloading latest version for adafruit/Adafruit_CircuitPython_Bundle.
```

It looks like its an undeclared runtime dependency that someone starting from scratch is going to hit.  Can setuptools be added to the requirements?

dhalbert, 2023-11-10:
See #143 for a generalization of the issue here.

ajs256, 2024-05-30:
I installed circup with `pipx`, as is recommended by pip when you're using a Homebrew Python, and needed to run `pipx inject circup setuptools` to get circup to work.

Just leaving this here for future readers.