adafruit/circup #144: Implement platform_system in requirements.txt

Neradoc, 2022-04-20:
Pip uses the following notation for platform-specific modules:
```
pywin32;platform_system=='Windows'
pyudev;platform_system=='Linux'
```
`circup install -r` would pick those as `pywin32;platform_system`, etc.

Would it be a good idea to:
- have circup find and read the platform_system parameter (and ignore the module)
- use a `platform_system=='circuitpython'` to specify modules that pip will ignore, but circup still install

This would allow pip and circup requirements to live in the same file to some extent for, say a project that is to be deployed on both a host and a target board. (For example circup itself can be a requirement for the host). Modules without a platform_system specified would still be installed (or ignored if known) as usual.

chabala, 2022-04-21:
Thank you for writing this up @Neradoc

Neradoc, 2022-04-21:
Interestingly since circup cuts the line at the `==` sign to ignore the version, this format works to have the module ignored by pip and installed by circup.
```
adafruit_datetime==0;platform_system=='circuitpython'
```