adafruit/circup #146: UnboundLocalError: local variable 'loc' referenced before assignment

bablokb, 2022-06-18:
Running circup fails because a variable is not assigned. Seems like an issue with the upcoming change in the mpy-format. In the logic before line 536 an additional else is missing.

    circup --path src install asyncio
    Found device at src, running CircuitPython 7.3.0.
    Searching for dependencies for: ['asyncio']
    Traceback (most recent call last):
    ...
    File "/usr/lib/python3.6/site-packages/circup/__init__.py", line 536, in extract_metadata
      if loc > -1:
    UnboundLocalError: local variable 'loc' referenced before assignment

bablokb, 2022-06-18:
Hm, I just had a look at the mpy-files and noticed that they are not in binary format but only pretend to be. So this seems also like a packaging error of the bundle-files and therefore circup can't detect the magic-bytes at the beginning of the files.

Nevertheless the code does not catch this issue correctly.

Neradoc, 2022-06-18:
Yes, it's a bug with the bundle, and an issue with circup not correctly reporting the error.
I did look a the circup bug, but that requires quite a bit of work to make sure that the error is reported in the right place.

Neradoc, 2022-06-19:
FYI the bug with the bundle is fixed (the Adafruit Bundle's latest release is good, the Community Bundle should be updated tonight).
The problem with circup crashing on bad mpy files is still there though, so that issue should remain open.

pingswept, 2022-06-19:
I ran into this too yesterday. mpy files that were not inside directories shipped as raw Python code, so the version information was undetectable, so the variable [loc](https://github.com/adafruit/circup/blob/main/circup/__init__.py#L536) was undefined.

A suggested fix would be to wrap the section linked above in a try/except, where the except clause is just `print("Can't find version information in current MPY file being examined.")

PR #155 (2022-12-09): Fix crash when bad MPY file
This avoids crashing into an exception when an mpy file is not found to be valid. That happens when it doesn't start with the expected headers (M3 or C5). This can happen if the file is corrupted, empty, or for example a py file is renamed mpy.
The version is set to a special value, and a warning is emitted during update similarly to when the format is valid but does not match the version of Circuitpython.
I didn't add a unit test yet.

fixes #146
closed 2022-12-09 by dhalbert