adafruit/circup #150: 'list' command should show installed module versions, regards of if they are out of date

chabala, 2022-07-21:
The list command as defined right now only provides useful output if it finds out of date modules. If I want to see the list of installed modules and what versions they are, that isn't available. Using the `--verbose` flag doesn't help, but it does add a lot of noise.

Further, if I remove all my modules and reinstall from requirements.txt, circup doesn't say what version is being installed, e.g. from README:

> $ circup install -r requirements.txt
> Installed 'adafruit_bmp280'.
> Installed 'adafruit_lis3mdl'.
> Installed 'adafruit_lsm6ds'.
> Installed 'adafruit_sht31d'.
> Installed 'neopixel'.

We assume it's the current version from the bundle, but it's difficult to verify.

Feature request: list should list all installed modules, and their versions; install should show version being installed.

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Ubuntu 21.10 x86_64

- Python version (run `python -version` or `python3 -version`):  Python 3.9.7

- Error message you are receiving, including any Python exception traces:  

> Found device at CIRCUITPY, running CircuitPython 7.3.2.
> All modules found on the device are up to date.

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): `$ circup --path CIRCUITPY list`

askpatrickw, 2022-07-22:
FYI : `circup freeze` lists all the modules and versions currently on the board.

Neradoc, 2022-07-22:
Note that the way freeze works is similar to `pip freeze`, it is reasonable not to create a file by default in my opinion, but I get that the format is not as good to read as the tabulated list of `list`.

On the other hand `circup list` does not follow `pip list`, and is specifically comparing versions, but I guess we could have something like a --all option. I like the idea of having circup help fully diagnose the status of the libraries.
Something like that maybe ?
```
Module             Version  Latest  Update Status  
------------------ -------- ------- -------------- 
adafruit_pixelbuf  1.1.5    1.1.5   Up To Date     
neopixel           6.3.0    6.3.3   Minor Version 
```

> Further, if I remove all my modules and reinstall from requirements.txt, circup doesn't say what version is being installed

Well you could run freeze to look at the versions afterward, but we could add the version in the install verbose print, like it is in the update verbose print. Or since the verbose flag is really verbose, we can add it in the normal run of install and update.
Something like that:
```
'adafruit_pixelbuf' (1.1.5) is already installed.
Installed 'neopixel' (6.3.3).
Update 'neopixel'? (6.3.2 to 6.3.3) [y/N]: 
```

todbot, 2022-07-29:
I too would like something akin to a `circup list --all` or `circup list --installed`.