adafruit/circup #152: `circup install -a` doesn't notice asyncio is needed

jepler, 2022-08-08:
I have a code.py with the content (in part):
```py
import asyncio
```

when I run `circup install -a`, it doesn't check for asyncio:
```
Found device at /media/jepler/CIRCUITPY, running CircuitPython 8.0.0-alpha.1-104-gf1826b054-dirty.
Searching for dependencies for: ['adafruit_display_text', 'adafruit_displayio_ssd1306']
Ready to install: ['adafruit_bitmap_font', 'adafruit_display_text', 'adafruit_displayio_ssd1306']

'adafruit_bitmap_font' is already installed.
'adafruit_display_text' is already installed.
'adafruit_displayio_ssd1306' is already installed.
```
As the name of this module doesn't follow the usual standard, it probably needs to be special-cased

Neradoc, 2022-08-08:
Can't reproduce, circup install -a doesn't care about special cases, the issue arises with pip names, like `adafruit-circuitpython-neopixel` that doesn't translate to `neopixel` on its own. So it happens when looking for dependencies in requirements.txt.

```
❯ circup --path /Volumes/FUNHOUSE install -a
Version 1.1.1.dev1+g9fa9554.d20220619 of circup is outdated. Version 1.1.2 was released Tuesday July 05, 2022.
Found device at /Volumes/FUNHOUSE, running CircuitPython 8.0.0-alpha.1-103-g0ed84486f.
Searching for dependencies for: ['asyncio', 'neopixel']
Ready to install: ['adafruit_pixelbuf', 'adafruit_ticks', 'asyncio', 'neopixel']

'adafruit_pixelbuf' is already installed.
'adafruit_ticks' is already installed.
'asyncio' is already installed.
'neopixel' is already installed.
```

jepler, 2022-08-09:
I can't reproduce this now either. Sorry for the noise.

jepler, 2022-08-09:
I set up a test folder like so
```
jepler@rat:~/circup-test$ grep . boot_out.txt code.py 
boot_out.txt:Adafruit CircuitPython 8.0.0-alpha.1-125-g0ae129951-dirty on 2022-08-08; Espressif ESP32-EYE with ESP32
boot_out.txt:Board ID:espressif_esp32_eye
boot_out.txt:UID:8EF9D6556A8A
code.py:import asyncio
```
and ran circup on it:
```
$ circup --path . install -a
Found device at ., running CircuitPython 8.0.0-alpha.1-125-g0ae129951-dirty.
Searching for dependencies for: ['asyncio']
Ready to install: ['adafruit_ticks', 'asyncio']

Installed 'adafruit_ticks'.
Installed 'asyncio'.
```

closed 2022-08-09 by jepler