adafruit/circup #154: pip install reports warning for upcoming pip 23.1

jliu70, 2022-10-27:
When I ran pip install circup, I saw the following WARNING message:

  DEPRECATION: circup is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559

Just wanted to bring this to the attention of the maintainers.

tekktrik, 2022-10-31:
It's probably worth solving this by switching to `pyproject.toml`.  A quick look at `setup.py` doesn't actually reveal any runtime logic that can't be implemented that way.

tekktrik, 2025-08-07:
pyproject.toml implemented!

closed 2025-08-07 by tekktrik