adafruit/circup #156: support web based workflow

vladak, 2022-11-15:
It would be nice if circup can work for microcontrollers that support only web based workflow. My (naive?) thought is: if a browser can list/upload/download files then why not circup.

anecdata, 2023-02-18:
Web Workflow File REST API
https://docs.circuitpython.org/en/latest/docs/workflows.html#file-rest-api

vladak, 2023-02-18:
Thanks, it seems that the REST API can be used to make circup to work with web workflow after a bit of hacking, at least for some initial commands:
```
$ python3 c.py --host fusebox --password xxx list
Found device at http://:xxx@fusebox, running CircuitPython 8.0.2.
The following modules are out of date or probably need an update.
Major Updates may include breaking changes. Review before updating.
MPY Format changes from Circuitpython 6 to 7 require an update.

Module             Version  Latest  Update Reason  
------------------ -------- ------- -------------- 
adafruit_ahtx0     1.0.16   1.0.17  Minor Version  
adafruit_pixelbuf  1.1.8    2.0.0   Major Version  
neopixel           6.3.6    6.3.8   Minor Version  
```

vladak, 2023-02-20:
I have most of the code ready for testing, however I cannot test with my ESP32 V2 since it [stops responding on port 80 after a file is uploaded](https://forums.adafruit.com/viewtopic.php?t=198974) and I have no simple means of restarting it. I ordered new ESP32 V2; once it arrives I will polish/debug the changes and submit a PR.

AstraLuma, 2023-10-14:
hey @vladak would you be interested in submitting your work as a PR?

vladak, 2023-10-14:
> hey @vladak would you be interested in submitting your work as a PR?

The PR #163 was submitted quite a long time ago.

commit 8b413ffe (2024-03-05): add support for web workflow

fixes #156
closed 2024-03-05 by tannewt