adafruit/circup #162: Circup crashes if local code.py present

todbot, 2023-02-12:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  MacOS 13.2 (Ventura)

- Python version (run `python -version` or `python3 -version`):  3.10.8

- Error message you are receiving, including any Python exception traces:  
```
 circup --help
Traceback (most recent call last):
  File "/opt/homebrew/bin/circup", line 5, in <module>
    from circup import main
  File "/opt/homebrew/lib/python3.10/site-packages/circup/__init__.py", line 22, in <module>
    import findimports
  File "/opt/homebrew/lib/python3.10/site-packages/findimports.py", line 84, in <module>
    import doctest
  File "/opt/homebrew/Cellar/python@3.10/3.10.8/Frameworks/Python.framework/Versions/3.10/lib/python3.10/doctest.py", line 100, in <module>
    import pdb
  File "/opt/homebrew/Cellar/python@3.10/3.10.8/Frameworks/Python.framework/Versions/3.10/lib/python3.10/pdb.py", line 77, in <module>
    import code
  File "/Users/tod/projects/synth/picotouch/circuitpython/picotouch/code.py", line 17, in <module>
    import board
  File "/opt/homebrew/lib/python3.10/site-packages/board.py", line 279, in <module>
    raise NotImplementedError(
NotImplementedError: Adafruit-PlatformDetect version 3.27.1 was unable to identify the board and/or microcontroller running the Darwin platform. Please be sure you have the latest packages running: 'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'
```

- List the steps to reproduce the problem below (if possible attach code or commands
  to run):  
Run `circup` when you happen to be in a directory with a `code.py`

I spent a while trying to figure out why `circup` was failing with weird long error with something about `NotImplementedError: Adafruit-PlatformDetect` on my Mac. Turns out I happened to be in a directory with a `code.py` and `circup` was parsing it? 

If this is designed behavior, is there a way to turn off this behavior?  I often run `circup` from a project source tree.

Neradoc, 2023-02-13:
`code` is a standard library in python, it is used by `pdb` (used by doctest, used by findimports), which then would be importing `code.py` instead. Since code.py contains probably `import board`, that would then run blinka.

However I was not able to reproduce that error. I use pyenv (using homebrew's installed python has issues) and have circup installed in its own environment. I can reproduce the error by importing pdb from the python REPL, but I can't make circup do it, I even tried installing and running circup in a local venv.
```
❯ python
Python 3.10.8 (main, Feb 13 2023, 01:30:05) [Clang 14.0.0 (clang-1400.0.29.202)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pdb
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/****/.pyenv/versions/3.10.8/lib/python3.10/pdb.py", line 77, in <module>
    import code
  File "/Users/****/ProjetsDev/Arduinos/macropad/MACROPAD-1/code.py", line 1, in <module>
    import board
ModuleNotFoundError: No module named 'board'
>>> 
```
```
Arduinos/macropad/MACROPAD-1 on  main [✘!?] via 🐍 v3.10.8 (venv) 
❯ which circup
/Users/****/ProjetsDev/Arduinos/macropad/MACROPAD-1/venv/bin/circup

Arduinos/macropad/MACROPAD-1 on  main [✘!?] via 🐍 v3.10.8 (venv)
❯ circup --help
Usage: circup [OPTIONS] COMMAND [ARGS]...

  A tool to manage and update libraries on a CircuitPython device.
[...]
```
Note that running python tools in a virtual environment has issues too, sometimes it fails because the modules that the tool uses are not installed in the venv, so I tend to run circup (via discotool) in a new window anyway.
I don't know what causes python to look for a local file when running a non-local script.

todbot, 2023-02-13:
Ahh, I see the issue.  I had `PYTHONPATH=.` env var set for unrelated testing in that window.
Unsetting that and the issue is resolved.  Sorry for the bother.

Closing this issue as the problem was between chair and keyboard.

closed 2023-02-13 by todbot