adafruit/circup #165: Library gets updated even if update not needed.

b-blake, 2023-03-20:
Hello,

Updated adafruit_vl53l1x
Updated adafruit_displayio_layout

The above libraries get updated each time I run circup.exe on a board that has them.  They get updated if they were last updated 5 minutes ago or 5 weeks ago.  I have attached the output below of the batch file that with one click I can update 8 AfaFruit CircuitPython projects.  The same script was run just a few minutes before.

Bruce

`Do you want to upgrade? [y/N]: y
Do all of the YES things here!
 Volume in drive C is Windows SSD
 Volume Serial Number is B984-5BB6

 Directory of C:\Users\Bruce\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts

12/04/2022  01:22 PM    <DIR>          .
12/04/2022  01:22 PM    <DIR>          ..
11/13/2022  06:11 PM             1,082 circup-script.py
11/13/2022  06:11 PM            74,752 circup.exe
01/26/2023  06:41 PM             1,817 circup.txt
11/13/2022  06:11 PM           107,948 findimports.exe
11/13/2022  06:11 PM           107,982 normalizer.exe
11/13/2022  06:37 PM           107,959 pip.exe
11/13/2022  06:37 PM           107,959 pip3.10.exe
11/13/2022  06:37 PM           107,959 pip3.exe
11/13/2022  06:11 PM           107,943 pysemver.exe
               9 File(s)        725,401 bytes
               2 Dir(s)  588,923,936,768 bytes free

Waiting for 0 seconds, press a key to continue ...
"WaveShare"
Version 1.1.3 of circup is outdated. Version 1.1.4 was released Saturday December 10, 2022.
Found device at H:, running CircuitPython 8.0.3.
A newer version of CircuitPython (8.0.4) is available.
Get it here: https://circuitpython.org/board/waveshare_rp2040_zero
None of the modules found on the device need an update.
"Maker 800"
Version 1.1.3 of circup is outdated. Version 1.1.4 was released Saturday December 10, 2022.
Found device at I:, running CircuitPython 8.0.4.
Found 1 module[s] needing update.
Updated ansi_escape_code
"Feather IR"
Version 1.1.3 of circup is outdated. Version 1.1.4 was released Saturday December 10, 2022.
Found device at J:, running CircuitPython 8.0.3.
A newer version of CircuitPython (8.0.4) is available.
Get it here: https://circuitpython.org/board/adafruit_feather_rp2040
Found 1 module[s] needing update.
Updated adafruit_displayio_layout
"TrinKey"
Version 1.1.3 of circup is outdated. Version 1.1.4 was released Saturday December 10, 2022.
Found device at K:, running CircuitPython 8.0.3.
A newer version of CircuitPython (8.0.4) is available.
Get it here: https://circuitpython.org/board/adafruit_qt2040_trinkey
Found 2 module[s] needing update.
Updated adafruit_vl53l1x
Updated adafruit_displayio_layout
"ESP32S2-SCR"
Version 1.1.3 of circup is outdated. Version 1.1.4 was released Saturday December 10, 2022.
Found device at L:, running CircuitPython 8.0.3.
A newer version of CircuitPython (8.0.4) is available.
Get it here: https://circuitpython.org/board/adafruit_qtpy_esp32s2
None of the modules found on the device need an update.
"Connect IR"
Version 1.1.3 of circup is outdated. Version 1.1.4 was released Saturday December 10, 2022.
Found device at M:, running CircuitPython 8.0.3.
A newer version of CircuitPython (8.0.4) is available.
Get it here: https://circuitpython.org/board/arduino_nano_rp2040_connect
Found 1 module[s] needing update.
Updated adafruit_displayio_layout
"TFT Display"
Version 1.1.3 of circup is outdated. Version 1.1.4 was released Saturday December 10, 2022.
Found device at N:, running CircuitPython 8.0.4.
Found 3 module[s] needing update.
Updated adafruit_vl53l1x
WARNING: Library font_maker repo has incorrect __version__
        metadata. Circup will assume it needs updating.
        Please file an issue in the library repo.
Updated font_maker
Updated adafruit_displayio_layout
"QT PY OLED"
Version 1.1.3 of circup is outdated. Version 1.1.4 was released Saturday December 10, 2022.
Found device at O:, running CircuitPython 8.0.3.
A newer version of CircuitPython (8.0.4) is available.
Get it here: https://circuitpython.org/board/adafruit_qtpy_rp2040
Found 2 module[s] needing update.
Updated adafruit_vl53l1x
Updated adafruit_displayio_layout
"No Drive"
Usage: circup [OPTIONS] COMMAND [ARGS]...
Try 'circup --help' for help.

Error: Invalid value for '--path': Directory 'P:' does not exist.
"No Drive"
Usage: circup [OPTIONS] COMMAND [ARGS]...
Try 'circup --help' for help.

Error: Invalid value for '--path': Directory 'Q:' does not exist.
"No Drive"
Usage: circup [OPTIONS] COMMAND [ARGS]...
Try 'circup --help' for help.

Error: Invalid value for '--path': Directory 'R:' does not exist.
"No Drive"
Usage: circup [OPTIONS] COMMAND [ARGS]...
Try 'circup --help' for help.

Error: Invalid value for '--path': Directory 'S:' does not exist.
 Volume in drive C is Windows SSD
 Volume Serial Number is B984-5BB6

 Directory of C:\Users\Bruce\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts

12/04/2022  01:22 PM    <DIR>          .
12/04/2022  01:22 PM    <DIR>          ..
11/13/2022  06:11 PM             1,082 circup-script.py
11/13/2022  06:11 PM            74,752 circup.exe
01/26/2023  06:41 PM             1,817 circup.txt
11/13/2022  06:11 PM           107,948 findimports.exe
11/13/2022  06:11 PM           107,982 normalizer.exe
11/13/2022  06:37 PM           107,959 pip.exe
11/13/2022  06:37 PM           107,959 pip3.10.exe
11/13/2022  06:37 PM           107,959 pip3.exe
11/13/2022  06:11 PM           107,943 pysemver.exe
               9 File(s)        725,401 bytes
               2 Dir(s)  588,919,992,320 bytes free
Press any key to continue . . .`

dhalbert, 2023-03-20:
Please update to the latest circup, though I don't think that may fix this. There may be something wrong with those two libraries. How about removing everything in `lib/` and then reinstalling, and then try updating again?

jposada202020, 2023-03-20:
@dhalbert some of the libraries were missing the __version__ information, so circup incorrectly reported. However, I thought this https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_Layout/pull/79 would have fixed the issue with displayio_layout.

@b-blake could you check again, during my tests once displayio_layout was updated to the newest version, circup did not complain. tks :)

b-blake, 2023-03-20:
At least two of the MCU have new libraries.  When I start a new project I get the current library bundle and copy the entire thing into the lib folder (if space available).  Being that the batch output says it was updated, I trust that it was updated.

I will look at updating to 1.1.4 from 1.1.3.  I would expect that if the issue I am seeing had been fixed in 1.1.4 it would be listed in the 'bug fixes' list.  Off I go to find 1.1.4.

Bruce

b-blake, 2023-03-20:
Hi,

I downloaded todays 20230320 compressed full library.  I copied the entire thing into J:, K:, and M:.  I then ran my script twice.  The result is below.  There are both new and old files that get a second probably unnecessary update.

Found 1 module[s] needing update.
Updated ansi_escape_code

Found 3 module[s] needing update.
Updated adafruit_vl53l1x
WARNING: Library font_maker repo has incorrect __version__
        metadata. Circup will assume it needs updating.
        Please file an issue in the library repo.
Updated font_maker
Updated adafruit_displayio_layout

Found 2 module[s] needing update.
Updated adafruit_vl53l1x
Updated adafruit_displayio_layout

etc.

I have not found an update file for 1.1.4.  What am I missing?

Bruce

`Do you want to upgrade? [y/N]: y
Do all of the YES things here!
 Volume in drive C is Windows SSD
 Volume Serial Number is B984-5BB6

 Directory of C:\Users\Bruce\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts

12/04/2022  01:22 PM    <DIR>          .
12/04/2022  01:22 PM    <DIR>          ..
11/13/2022  06:11 PM             1,082 circup-script.py
11/13/2022  06:11 PM            74,752 circup.exe
01/26/2023  06:41 PM             1,817 circup.txt
11/13/2022  06:11 PM           107,948 findimports.exe
11/13/2022  06:11 PM           107,982 normalizer.exe
11/13/2022  06:37 PM           107,959 pip.exe
11/13/2022  06:37 PM           107,959 pip3.10.exe
11/13/2022  06:37 PM           107,959 pip3.exe
11/13/2022  06:11 PM           107,943 pysemver.exe
               9 File(s)        725,401 bytes
               2 Dir(s)  588,669,685,760 bytes free

Waiting for 3 seconds, press a key to continue ...
"WaveShare"
Version 1.1.3 of circup is outdated. Version 1.1.4 was released Saturday December 10, 2022.
Found device at H:, running CircuitPython 8.0.3.
A newer version of CircuitPython (8.0.4) is available.
Get it here: https://circuitpython.org/board/waveshare_rp2040_zero
None of the modules found on the device need an update.
"Maker 800"
Version 1.1.3 of circup is outdated. Version 1.1.4 was released Saturday December 10, 2022.
Found device at I:, running CircuitPython 8.0.4.
Found 1 module[s] needing update.
Updated ansi_escape_code
"Feather IR"
Version 1.1.3 of circup is outdated. Version 1.1.4 was released Saturday December 10, 2022.
Found device at J:, running CircuitPython 8.0.3.
A newer version of CircuitPython (8.0.4) is available.
Get it here: https://circuitpython.org/board/adafruit_feather_rp2040
Found 3 module[s] needing update.
Updated adafruit_vl53l1x
WARNING: Library font_maker repo has incorrect __version__
        metadata. Circup will assume it needs updating.
        Please file an issue in the library repo.
Updated font_maker
Updated adafruit_displayio_layout
[... 62 lines cut ...]
Error: Invalid value for '--path': Directory 'R:' does not exist.
"No Drive"
Usage: circup [OPTIONS] COMMAND [ARGS]...
Try 'circup --help' for help.

Error: Invalid value for '--path': Directory 'S:' does not exist.
 Volume in drive C is Windows SSD
 Volume Serial Number is B984-5BB6

 Directory of C:\Users\Bruce\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts

12/04/2022  01:22 PM    <DIR>          .
12/04/2022  01:22 PM    <DIR>          ..
11/13/2022  06:11 PM             1,082 circup-script.py
11/13/2022  06:11 PM            74,752 circup.exe
01/26/2023  06:41 PM             1,817 circup.txt
11/13/2022  06:11 PM           107,948 findimports.exe
11/13/2022  06:11 PM           107,982 normalizer.exe
11/13/2022  06:37 PM           107,959 pip.exe
11/13/2022  06:37 PM           107,959 pip3.10.exe
11/13/2022  06:37 PM           107,959 pip3.exe
11/13/2022  06:11 PM           107,943 pysemver.exe
               9 File(s)        725,401 bytes
               2 Dir(s)  588,666,441,728 bytes free
Press any key to continue . . .`

dhalbert, 2023-03-20:
[Use triple backticks on their own lines to mark multiple lines of code. (Edit your posts above and see what I did.)]

To upgrade `circup`, use `pip3 install --upgrade circup` (or maybe `pip`, depending on your particular Python installation).

What are `J:`, `K:`, and `M:`? Are these your ESP32-nn boards?

If you just have one board attached, do you see the same problem?

dhalbert, 2023-03-20:
Since it seems confined to
```
Updated adafruit_vl53l1x
Updated adafruit_displayio_layout
```
I think this may be a library problem, not a circup problem.

jposada202020, 2023-03-21:
ill do some test on displayio this week to see what is going on..

b-blake, 2023-03-21:
Hi,

If I update one device I get the same result that device gave before.

**ansi_escape_code** gets updated and may be a clue.

Bruce

```
J:
Adafruit CircuitPython 8.0.3 on 2023-02-23; Adafruit Feather RP2040 with rp2040
Board ID:adafruit_feather_rp2040
UID:38503039350B2620
boot.py output:
	Write to Flash  : False
	Write to SD Card: True
	boot.py Finis

K:
Adafruit CircuitPython 8.0.3 on 2023-02-23; Adafruit QT2040 Trinkey with rp2040
Board ID:adafruit_qt2040_trinkey
UID:DF60481763303039
boot.py output:
	Write to SD Card: True
	Write to Flash  : False
	boot.py Finis

M:
Adafruit CircuitPython 8.0.3 on 2023-02-23; Arduino Nano RP2040 Connect with rp2040
Board ID:arduino_nano_rp2040_connect
UID:5031503337360005
boot.py output:
	Write to Flash  : False
	Write to SD Card: True
	boot.py Finis```

jposada202020, 2023-03-21:
@b-blake circup 1.1.15 (just released) will solve the issue for the displayio_layout. I have tested the changes. For the VL5311x library I have made a PR, once it is merged this will be fixed https://github.com/adafruit/Adafruit_CircuitPython_VL53L1X/pull/15

b-blake, 2023-03-21:
Hi,

I updated to 1.1.5.  We now only have two libraries to beat into submission by my reckoning.  They may already be pinned to the mat in 1.1.6.  Second run at: 11:45 AM EDT.

ansi_escape_code.mpy
font_maker.mpy

The results below are from a run less than 5 minutes after the first run for today, both runs with version 1.1.5.

Bruce
```
"I: Maker 800"
Found device at I:, running CircuitPython 8.0.4.
Found 1 module[s] needing update.
Updated ansi_escape_code

"J: Feather IR"
Found device at J:, running CircuitPython 8.0.3.
A newer version of CircuitPython (8.0.4) is available.
Get it here: https://circuitpython.org/board/adafruit_feather_rp2040
Found 1 module[s] needing update.
WARNING: Library font_maker repo has incorrect __version__
        metadata. Circup will assume it needs updating.
        Please file an issue in the library repo.
Updated font_maker
```

jposada202020, 2023-03-21:
Hello That is good, two things, I have made a PR for the font_maker.py here https://github.com/adafruit/Adafruit_CircuitPython_GFX/pull/32

However, the Ansi_escape_code is a community library, and is supported by community members. As mentioned in the error message, the issue should be submitted to the Library Repo, and not in the circup library. 

I have submitted a PR in the community library https://github.com/s-light/CircuitPython_ansi_escape_code/pull/2, but it will depend on the library owner, when the PR will be merged. I recommend following this PR, for this :)

b-blake, 2023-03-21:
> here adafruit/Adafruit_CircuitPython_GFX#32
> s-light/CircuitPython_ansi_escape_code#2

J,

Thank you for doing the PR for both items.  "community library" did not
ring a bell for me to know to go to the community library to file the
issue.  Not knowing the AdaFruit organizational structure it was a
difference without a difference.  I promise you I will not be any smarter
next time, but I will try.

Bruce

P.S.  FYI - I am not the young sports car you think I am.
This chassis was built in 1947 and has had major repairs, and the paint is
flaking off ;-)

jposada202020, 2023-03-21:
No worries :). 1947 seems like a good year :). FYI the community libraries are libraries maintained by community members, for a complete list of the 100: https://github.com/adafruit/CircuitPython_Community_Bundle/blob/main/circuitpython_community_library_list.md maybe you will find something that you could use in the future. Cheers :)

b-blake, 2024-02-14:
Hello,

I just ran CIRCUP twice on the same device with 9.0.0 Beta 0 loaded.  All 330 library files were updated both times.
I assume the first run updated all the library files to the compatible format.
The second run should have recognized that library files had the corrected format and not updated.

Do I need to wait for something else to be updated before I try again?

dhalbert, 2024-02-14:
You need to update to the 1.5.3 release, which I created a few minutes ago. So do `pip3 install --upgrade circup` and then try again.

b-blake, 2024-02-14:
Updated and it works.
Because the thread was closed I thought the update had been done.
I apologize for jumping the gun.

PR #198 (2024-02-13): Handle mpy version checking for Circuitpython 9
Fixes #165.
Fixes #196.

- Handle .mpy files with bytecode version `C\x06`. The `__version__` string is not findable in as straightforward a way because the way dictionary entries are in the .mpy has changed, so just look for the first string that looks like a version number.

Tested on 8.2.9 and 9.0.0-beta.0, including upgrading and downgrading between those versions.