adafruit/circup #169: can't install 1.2.3

dsssssssss9, 2023-06-06:
- Platform/operating system : Ubuntu 20.04

- Python version : Python 3.8.10

- Error message you are receiving, : Requirement already satisfied: circup in ./.local/lib/python3.8/site-packages (1.2.2)

- List the steps to reproduce the problem below

pip3 install circup

even though 1.2.3has been released every time i try to install it i get the above error

Even happens if i specify the version to install with...
python3 -m pip install "circup==1.2.3"

Any help appreciated

dhalbert, 2023-06-06:
I would expect `pip3 install -upgrade circup` to upgrade. I just tried it myself on Ubuntu 22.04, and it worked. But you are on an older version. The change between 1.2.2 and 1.2.3 was to upgrade `requests`, which says it still supports 3.7 and up. If this still doesn't work for you, what does `pip3 list` show? I'll compare it with what I have, and let's compare it with https://github.com/adafruit/circup/blob/main/requirements.txt.

dsssssssss9, 2023-06-06:
Hi

Thanks for getting back to me so quickly ...

here are my results ....

`pip3 list
Package                                 Version
--------------------------------------- --------------------
Adafruit-Blinka                         6.15.0
adafruit-board-toolkit                  1.1.0
adafruit-circuitpython-busdevice        5.1.1
adafruit-circuitpython-displayio-sh1106 1.2.3
adafruit-circuitpython-max7219          1.4.1
adafruit-circuitpython-us100            1.1.5
Adafruit-GPIO                           1.0.3
adafruit-nrfutil                        0.5.3.post16
Adafruit-PlatformDetect                 3.16.0
Adafruit-PureIO                         1.1.8
Adafruit-SSD1306                        1.6.2
appdirs                                 1.4.3
apturl                                  0.5.2
argcomplete                             1.12.3
astroid                                 2.5.2
asttokens                               2.0.4
backcall                                0.2.0
bcrypt                                  3.1.7
black                                   20.8b1
blinker                                 1.4
Brlapi                                  0.7.0
cajarename                              19.7.15
certifi                                 2019.11.28
cffi                                    1.15.0
chardet                                 3.0.4
circup                                  1.2.2
click                                   8.0.3
colorama                                0.4.3
colorzero                               1.1
command-not-found                       0.3
configobj                               5.0.6
cryptography                            35.0.0
cupshelpers                             1.0
cycler                                  0.10.0
dbus-python                             1.2.16
decorator                               4.4.2
defer                                   1.0.6
deja-dup-caja                           0.0.6
distlib                                 0.3.0
distro                                  1.4.0
distro-info                             0.23ubuntu1
dnspython                               1.16.0
docutils                                0.16
duplicity                               0.8.12.0
ecdsa                                   0.17.0
entrypoints                             0.3
fasteners                               0.14.1
filelock                                3.0.12
findimports                             2.1.0
flake8                                  3.9.0
folder-color-caja                       0.0.86
folder-color-common                     0.0.86
future                                  0.18.2
gpg                                     1.13.1-unknown
gpiozero                                1.5.1
grove.py                                0.6
guizero                                 1.1.1
httplib2                                0.14.0
idna                                    2.8
importlib-metadata                      1.5.0
intelhex                                2.3.0
[... 110 lines cut ...]
tomli                                   1.2.3
tornado                                 6.1
traitlets                               5.0.5
typed-ast                               1.4.2
typing-extensions                       3.7.4.3
ubittool                                0.5.0
ubuntu-advantage-tools                  8001
ubuntu-drivers-common                   0.0.0
uflash                                  1.2.0
ufw                                     0.36
unattended-upgrades                     0.1
update-checker                          0.18.0
urllib3                                 1.25.8
usb-creator                             0.3.7
userpath                                1.7.0
virtualenv                              20.0.17
wadllib                                 1.3.3
wcwidth                                 0.2.5
websocket-client                        1.2.1
websockets                              10.0
wheel                                   0.34.2
wrapt                                   1.12.1
xkit                                    0.0.0
zipp                                    1.0.0
`

dhalbert, 2023-06-06:
It appears to me `circup`'s `requirements.txt` is not being used at all. Both my packages and yours are behind the list, though yours are further behind than mine.

I am not sure why they are all pinned like that. That unnecessarily limits other packages.

Here are some instructions for updating packages: https://www.activestate.com/resources/quick-reads/how-to-update-all-python-packages/. Scroll down to "Update all Python Packages on Linux".

@Neradoc @tekktrik I am not that familiar with the packaging requirements issues here. Do you have a comment? Thanks.

dsssssssss9, 2023-06-06:
I have tried both sets of instructions on the site you kindly linked 

took ages & several errors scrolled by that unfortunate i did not record them

i have tried 
`pip3 uninstall  circup
Found existing installation: circup 1.2.2
Uninstalling circup-1.2.2:
  Would remove:
    /home/jkd/.local/bin/circup
    /home/jkd/.local/lib/python3.8/site-packages/circup-1.2.2.dist-info/*
    /home/jkd/.local/lib/python3.8/site-packages/circup/*
Proceed (Y/n)? Y
  Successfully uninstalled circup-1.2.2
jkd@jkd-mininuc:~$ circup
circup: command not found
`
followed  by

`python3 -m pip install "circup==1.2.3"
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple/
ERROR: Could not find a version that satisfies the requirement circup==1.2.3 (from versions: 0.0.1, 0.0.3, 0.0.4, 0.0.5, 0.0.6, 0.0.8, 0.1.0, 0.2.2, 0.3.1, 0.4.0, 0.4.1, 0.5.0, 0.6.0, 0.6.1, 0.7.0, 0.8.0, 0.9.0, 0.9.1, 0.9.2, 0.9.3, 0.9.4, 0.9.5, 0.9.7, 0.9.8, 0.9.9, 0.9.10, 0.9.11, 1.0.0, 1.0.1, 1.0.2, 1.0.3, 1.0.4, 1.1.0, 1.1.1, 1.1.2, 1.1.3, 1.1.4, 1.1.5, 1.2.0, 1.2.1, 1.2.2)
ERROR: No matching distribution found for circup==1.2.3
` 

so have installed 1.2.2 again for now 
Hope we can get to the bottom of this?

tekktrik, 2023-06-06:
I can take a look in a couple days, or hopefully tonight if I have a free moment!

dsssssssss9, 2023-06-06:
Latest results after doing the above....

`pip3 list
Package                                 Version
--------------------------------------- --------------------
Adafruit-Blinka                         8.19.0
adafruit-board-toolkit                  1.1.1
adafruit-circuitpython-busdevice        5.2.4
adafruit-circuitpython-displayio-sh1106 1.2.10
adafruit-circuitpython-framebuf         1.6.1
adafruit-circuitpython-max7219          1.5.9
adafruit-circuitpython-requests         1.13.2
adafruit-circuitpython-typing           1.9.2
adafruit-circuitpython-us100            1.1.13
Adafruit-GPIO                           1.0.3
adafruit-nrfutil                        0.5.3.post16
Adafruit-PlatformDetect                 3.45.2
Adafruit-PureIO                         1.1.10
Adafruit-SSD1306                        1.6.2
appdirs                                 1.4.4
apturl                                  0.5.2
argcomplete                             3.0.8
astroid                                 2.15.4
asttokens                               2.2.1
backcall                                0.2.0
bcrypt                                  4.0.1
black                                   23.3.0
blinker                                 1.6.2
Brlapi                                  0.7.0
cajarename                              19.7.15
capstone                                4.0.2
certifi                                 2023.5.7
cffi                                    1.15.1
chardet                                 5.1.0
charset-normalizer                      3.1.0
circup                                  1.2.2
click                                   8.1.3
cmsis-pack-manager                      0.5.2
colorama                                0.4.6
colorzero                               2.0
comm                                    0.1.3
command-not-found                       0.3
configobj                               5.0.8
contourpy                               1.0.7
cryptography                            40.0.2
cupshelpers                             1.0
cycler                                  0.11.0
dbus-python                             1.2.16
debugpy                                 1.6.7
decorator                               5.1.1
defer                                   1.0.6
deja-dup-caja                           0.0.6
dill                                    0.3.6
distlib                                 0.3.6
distro                                  1.8.0
distro-info                             1.0
dnspython                               2.3.0
docutils                                0.20
duplicity                               0.8.12.0
ecdsa                                   0.18.0
entrypoints                             0.4
executing                               1.2.0
fasteners                               0.18
filelock                                3.12.0
findimports                             2.3.0
flake8                                  6.0.0
folder-color-caja                       0.0.86
folder-color-common                     0.0.86
fonttools                               4.39.4
future                                  0.18.3
[... 137 lines cut ...]
tomlkit                                 0.11.8
tornado                                 6.3.1
traitlets                               5.9.0
typed-ast                               1.5.4
typing_extensions                       4.5.0
ubittool                                0.6.0
ubuntu-advantage-tools                  8001
ubuntu-drivers-common                   0.0.0
uflash                                  2.0.0
ufw                                     0.36
unattended-upgrades                     0.1
update-checker                          0.18.0
urllib3                                 2.0.2
usb-creator                             0.3.7
userpath                                1.8.0
virtualenv                              20.23.0
wadllib                                 1.3.6
wcwidth                                 0.2.6
websocket-client                        1.5.1
websockets                              11.0.3
wheel                                   0.40.0
wrapt                                   1.15.0
xkit                                    0.0.0
zipp                                    3.15.0
`

tekktrik, 2023-06-06:
Actually, looking at your error log it doesn't actually look like it's checking PyPI but some other index, did you or your organization set it up that way?

dsssssssss9, 2023-06-06:
organisation?
It's  just me at my dining room table i'm afraid!!

Installed circup as per instructions very  long time ago & until now the upgrade has always worked.

Just a home Ubuntu user & as far as i know nothing particular special about my setup / configuration 

not an earth shattering problem - but would love to find out what's causing it & how to solve it

tekktrik, 2023-06-06:
It looks like it's checking `https://pypi.tuna.tsinghua.edu.cn/simple/` for packages, which isn't the typical PyPI index URL.  Do you know why it's only checking there?

dsssssssss9, 2023-06-06:
have got 1.2.3 installed by doing  the following,,,,

edited the contents of ~/.pip/pip.conf

from 

`[global]^M
 timeout=6000 ^M
 index-url = https://pypi.tuna.tsinghua.edu.cn/simple/ ^M
 trusted-host = pypi.tuna.tsinghua.edu.cn`
 
 to
 
 `[global]
 timeout=6000
 index-url = https://pypi.python.org/simple
 trusted-host = https://pypi.python.org/      
`

uninstalled 1.2.2 as previous message

Installed via ...
`pip3 install --user circup`

1.2.3 sucessfully installs now

Have not got the faintest idea how the previous URL " https://pypi.tuna.tsinghua.edu.cn/simple" came to be the default index!!

Thank you @dhalbert  @tekktrik

tekktrik, 2023-06-06:
Glad it's been resolved!

closed 2023-06-06 by tekktrik