adafruit/circup #171: circup fails to find bundle?

todbot, 2023-06-24:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  MacOS Venture 13.4

- Python version (run `python -version` or `python3 -version`):  3.11.4

- Error message you are receiving, including any Python exception traces: 
   `There was a problem downloading the bundle. Please try again in a moment.`

  It looks like `circup` is trying to download `adafruit-circuitpython-bundle-py-20230624.zip` 
  instead of `adafruit-circuitpython-bundle-8.x-mpy-20230624.zip`

- List the steps to reproduce the problem below (if possible attach code or commands
  to run):  
```
% circup --verbose install adafruit_midi
Logging to /Users/tod/Library/Logs/circup/circup.log

06/24/2023 11:36:06 INFO: ### Started Circup ###
06/24/2023 11:36:06 INFO: Checking for a newer version of circup
06/24/2023 11:36:06 INFO: Found device: /Volumes/CIRCUITPY
06/24/2023 11:36:06 INFO: Requesting redirect information: https://github.com/adafruit/circuitpython/releases/latest
06/24/2023 11:36:06 INFO: Tag: '8.1.0'
Found device at /Volumes/CIRCUITPY, running CircuitPython 8.2.0-rc.0.
06/24/2023 11:36:06 INFO: Using bundles: adafruit/Adafruit_CircuitPython_Bundle, adafruit/CircuitPython_Community_Bundle, circuitpython/CircuitPython_Org_Bundle
06/24/2023 11:36:06 INFO: Checking library updates for adafruit/Adafruit_CircuitPython_Bundle.
06/24/2023 11:36:06 INFO: Requesting redirect information: https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/latest
06/24/2023 11:36:07 INFO: Tag: '20230624'
06/24/2023 11:36:07 INFO: New version available (20230624).
Downloading latest version for adafruit/Adafruit_CircuitPython_Bundle.

06/24/2023 11:36:07 INFO: Downloading bundle: https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/download/20230624/adafruit-circuitpython-bundle-py-20230624.zip
06/24/2023 11:36:08 WARNING: Unable to connect to https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/download/20230624/adafruit-circuitpython-bundle-py-20230624.zip
There was a problem downloading the bundle. Please try again in a moment.
06/24/2023 11:36:08 ERROR: 404 Client Error: Not Found for url: https://objects.githubusercontent.com/github-production-release-asset-2e65be/72229984/ad67165e-fd1d-43c4-801f-90124bbe8d79?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230624%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230624T183608Z&X-Amz-Expires=300&X-Amz-Signature=ba3f59ca9c3c3d84675cc2bbf07078aa2c5833567be12d0ccdb7889b6807ba54&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=72229984&response-content-disposition=attachment%3B%20filename%3Dadafruit-circuitpython-bundle-py-20230624.zip&response-content-type=application%2Foctet-stream
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.11/site-packages/circup/__init__.py", line 483, in ensure_latest_bundle
    get_bundle(bundle, tag)
  File "/opt/homebrew/lib/python3.11/site-packages/circup/__init__.py", line 702, in get_bundle
    r.raise_for_status()
  File "/opt/homebrew/lib/python3.11/site-packages/requests/models.py", line 1021, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://objects.githubusercontent.com/github-production-release-asset-2e65be/72229984/ad67165e-fd1d-43c4-801f-90124bbe8d79?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230624%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230624T183608Z&X-Amz-Expires=300&X-Amz-Signature=ba3f59ca9c3c3d84675cc2bbf07078aa2c5833567be12d0ccdb7889b6807ba54&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=72229984&response-content-disposition=attachment%3B%20filename%3Dadafruit-circuitpython-bundle-py-20230624.zip&response-content-type=application%2Foctet-stream
```

dhalbert, 2023-06-24:
I think this is a bundle problem. There was an error building the bundle last night. I will try to re-run it.

Neradoc, 2023-06-24:
Yeah the latest release is missing some files.
It ran once, and had an error that might be a temporary issue, but rerunning it lead to another error due to some files already existing in the assets apparently.
If rerunning it doesn't work, I would think deleting the last release could be a temporary solution ?

dhalbert, 2023-11-10:
Bundle fetching made more robust by #184.

closed 2023-11-10 by dhalbert