adafruit/circup #174: Can't handle UTF-8 in Windows 11 Home

djairjr, 2023-07-24:
Can't run Circup at Python 3.11 in windows 11 home.

```
circup --version
CircUp, A CircuitPython module updater. Version 1.2.3

circup --update
Found device at E:\, running CircuitPython 8.2.0.
There was a problem: 'utf-8' codec can't decode byte 0xeb in position 16384: invalid continuation byte
```

djairjr, 2023-07-24:
I am using a Raspberry Pi Pico W, and my Windows is Pt-BR (Brazilian Portuguese).
The libraies:
adafruit_ads1x15                                                     
adafruit_displayio_layout                                            
adafruit_pn532                                                       
adafruit_ht16k33                                                     
adafruit_minimqtt                                                    
adafruit_httpserver                                                  
asyncio                                                              
debouncer                                                            
adafruit_bitmap_font                                                 
adafruit_character_lcd                                               
adafruit_display_shapes                                              
adafruit_fancyled                                                    
ticks                                                                
pioasm                                                               
adafruit_register                                                    
simplemath                                                           
simple-text-display                                                  
adafruit_display_text                                                
adafruit_ili9341.mpy                                                 
adafruit_matrixkeypad.mpy                                            
adafruit_pcf8574.mpy                                                 
adafruit_pcf8575.mpy                                                 
adafruit_requests.mpy                                                                                                       
adafruit_ntp.mpy                                                     
morsecode.mpy                                                        
adafruit_datetime.mpy                                                
adafruit_debouncer.mpy                                               
adafruit_pcf8563.mpy                                                 
adafruit_ticks.mpy                                                   
neopixel.mpy

djairjr, 2023-07-24:
I just keep trying to update all libraries, and now, just work.
I will try with another board. But now, Circup works.

dhalbert, 2023-07-24:
> I just keep trying to update all libraries, and now, just work.

Glad it worked -- I have no explanation. It's possible there was a glitch reading from the drive, though that's very unusual. You might consider erasing and reformatting CIRCUITPY (after saving your work) to get rid of any corruption:
```py
import storage
storage.erase_filesystem()
```
I'll close this but please post again if it recurs, especially if it seems reproducible.

closed 2023-07-24 by dhalbert