adafruit/circup #181: How are dependencies with Community Bundle libraries listed in requirements.txt?

CedarGroveStudios, 2023-10-21:
see https://github.com/adafruit/CircuitPython_Community_Bundle/issues/192

tannewt, 2023-10-23:
I think @FoamyGuy is looking into this.

FoamyGuy, 2023-10-28:
@tannewt and @CedarGroveStudios I've done some poking around inside of circup to try to better understand how it operates and done some thinking about this functionality. 

The `pyproject.toml` files aren't currently included in the zipped bundles that circup downloads, so in order to go that route we would need to alter the tool that builds the bundles to have it include that file. If we do that perhaps it should also live inside of the requirements directory within the bundle. 

However I've also had another idea that will be very simple to implement. 

`requirements.txt` files allow comments with the hashtag, so we could make our own syntax that circup can look for within a comment. 

something like:
```
Adafruit-Blinka
# circup_requirement: cedargrove_midi_tools
```
This way pip will ignore that line. We can slightly modify the logic here: https://github.com/adafruit/circup/blob/abd225169a05d1e138e6e261ea76f89fb1093f8b/circup/__init__.py#L1082-L1084 to look for our special pre-fix comment and when found add that library to the list of requirements to get installed.

What do you think about that option?

FoamyGuy, 2023-10-30:
> requirements.txt is included in the bundle for each library but the toml file isn't?

That is correct. Here is an example of the structure of a bundle currently:

```
examples/
    74hc595_daisy_chain/
        74hc595_daisy_chain.py
    ...

lib/
    adafruit_ads1x15/
        [library .mpy files]
    ...
    [single file library .mpy files...]

requirements/
    adafruit_ads1x15/
        requirements.txt
    ....
README.txt
VERSIONS.txt
```
I'll look into the bundle builder to see about adding pyproject.toml to the bundles. @tannewt where should that file go inside the bundle? Inside of the existing `requirements/` directory? or make a new directory for it?

tannewt, 2023-10-31:
> @tannewt where should that file go inside the bundle? Inside of the existing `requirements/` directory? or make a new directory for it?

I don't have a preference.

closed 2023-11-21 by CedarGroveStudios