adafruit/circup #186: There was a problem downloading the bundle. Please try again in a moment

dsssssssss9, 2023-10-29:
- Platform/operating system 
Ubuntu Release 20.04.6 LTS (Focal Fossa) 64-bit

- Python version 
- Python 3.8.10

- Error message you are receiving,
Downloading latest version for adafruit/CircuitPython_Community_Bundle.

There was a problem downloading the bundle. Please try again in a moment.

- List the steps to reproduce the problem below
- run any command such as "circup list"

geoff-coppertop, 2023-10-29:
I looked at the output using --verbose and it's because there was a failure in the actions for the community bundle and no output was produced which circup is choking on when it makes its get request.

dhalbert, 2023-10-29:
I have a PR fix for this: #184. We are also fixing the bundle building right now.

dsssssssss9, 2023-10-29:
@geoff-coppertop 

Thank you for the information

dsssssssss9, 2023-10-29:
@dhalbert 

as always you are on the case !

Appreciate this

closed 2023-10-30 by FoamyGuy